Rahix/avr-hal #2: ATMega328p Support

jonahbron, 2019-06-22:
I would like to add support for the ATMega328p to this crate so that I can utilize embedded-hal drivers on my Arduino Uno.  Being super new to this, I'd really appreciate a high-level description of what that would take.  I'm going to fork this repo and inspect the other boards to figure out how, but a little direction could speed things up a lot.  Thanks!

Rahix, 2019-06-22:
Hey, thanks for your interest!

There are a few things that need to be done.  The first aren't actually in this repository, but over in [`avr-device`](https://github.com/Rahix/avr-device).

## 1. Add the MCU to `avr-device`
You are in luck!  This step is already done, because I anticipated this chip to need an implementation soon ;)  Just for completeness of this documentation, I'll include it.  Detailed instructions are over [here](https://github.com/Rahix/avr-device#adding-a-new-chip).

## 2. Add svd patches in `avr-device`
This is probably the first thing you need to start with.  Unfortunately, the atdf files that Microchip (former Atmel) supplies are not of great quality.  To fix them, we have included the svd patch tool from [stm32-rs](https://github.com/stm32-rs/stm32-rs) in our crate.  You need to make use of it to rename and change a few registers.

Luckily, the ATmega328P is very similar to the already implemented ATmega32U4.  This should mean that you can for the most part copy the patches from that chip.  You can find them [here](https://github.com/Rahix/avr-device/blob/master/patch/atmega32u4.yaml).  Documentation is in the [README](https://github.com/Rahix/avr-device#adding-a-new-chip).

---

Now, onto things in this repo:

## 3. Add a `chips/atmega328p-hal/` crate
Just follow what was done for the other chips.  Specifically [ATmega32U4](https://github.com/Rahix/avr-hal/tree/master/chips/atmega32u4-hal) because of its similarity.

There are a few important things:
* The rustc target description file.  In your case `avr-atmega328p.json`.  You can just copy the one from 32U4 again and just change the chipname inside it wherever it appears.  The rest can stay the same.
* Inside the sources for the existing chips, you will see a lot of macro calls.  These are used to create instances of certain peripherals.  The i2c (TWI) and UART ones will probably be almost exactly the same; the port definitions might be slightly different.  It should be fairly straightforward to get the necessary information from the datasheet.

## 4. Adding a board crate for your board
In your case, this is `boards/arduino-uno`.  You can use my [Arduino Leonardo](https://github.com/Rahix/avr-hal/tree/master/boards/arduino-leonardo) crate as a reference.  This crate is meant as a wrapper around the `hal` crate that has an interface which more closely resembles the actual hardware.  For example, there is a [wrapper](https://github.com/Rahix/avr-hal/blob/master/boards/arduino-leonardo/src/pins.rs) around the port peripherals which renames the pins to the names printed on the PCB.

---

If you have any questions, don't be afraid to ask!  Especially don't hesitate to report weird behaviors! There are still some compiler issues and I really wouldn't want you to suffer through them like I had to ...

Anyway, good luck!

jonahbron, 2019-06-23:
Thanks so much for the direction @Rahix.  There's clearly a lot here I don't know, but I'm looking forward to figuring it out.  If it's okay I'll keep this issue open as I continue working on this.

jonahbron, 2019-06-24:
Opened PR #3 that adds a chip crate for atmega328p.

jonahbron, 2019-06-24:
PR #3 now contains the board crate for Arduino Uno.

Rahix, 2019-06-27:
Support has landed in 1e1ed1b68e684d45442a60c19da79da839fcfe46. :tada:

jonahbron, 2019-06-27:
Woop, thanks for the help

closed 2019-06-27 by jonahbron