Rahix/avr-hal #4: Errors from ATMega32u4 chip

jonahbron, 2019-06-25:
While trying to build on my fork (which revealed some bugs in my code fortunately), I'm seeing errors come from the atmega32u4 chip crate.

```
error[E0412]: cannot find type `PORTE` in module `crate::atmega32u4`
  --> chips/atmega32u4-hal/src/port.rs:14:30
   |
14 |         E(crate::atmega32u4::PORTE, porte, pine),
   |                              ^^^^^ help: a struct with a similar name exists: `PORTB`

error[E0412]: cannot find type `PORTF` in module `crate::atmega32u4`
  --> chips/atmega32u4-hal/src/port.rs:15:30
   |
15 |         F(crate::atmega32u4::PORTF, portf, pinf),
   |                              ^^^^^ help: a struct with a similar name exists: `PORTB`

error[E0412]: cannot find type `PORTE` in module `atmega32u4`
  --> chips/atmega32u4-hal/src/port.rs:87:38
   |
87 |         impl PortExt for atmega32u4::PORTE {
   |                                      ^^^^^ help: a struct with a similar name exists: `PORTB`

error[E0412]: cannot find type `PORTF` in module `atmega32u4`
   --> chips/atmega32u4-hal/src/port.rs:103:38
    |
103 |         impl PortExt for atmega32u4::PORTF {
    |                                      ^^^^^ help: a struct with a similar name exists: `PORTB`

error[E0412]: cannot find type `USART1` in module `crate::atmega32u4`
 --> chips/atmega32u4-hal/src/usart.rs:5:40
  |
5 |         peripheral: crate::atmega32u4::USART1,
  |                                        ^^^^^^ help: a struct with a similar name exists: `USART0`

error: aborting due to 5 previous errors
```

Are  these something you'd like me to try to fix, or are these errors showing for some non-obvious user-error reason?

Rahix, 2019-06-25:
I am not able to reproduce these errors, it works perfectly fine for me ... From the error messages it looks like you are trying to build `atmega32u4-hal` with an `avr-device` crate for ATmega328P?  At least 32U4 has `PORT{E,F}` while 328P does not ...

Can you try cleaning your `target/` and rebuilding?

Rahix, 2019-06-25:
Oh, I see what you did ... You probably attempted to build from the repository root?  Please follow the exact instructions from the [README](https://github.com/jonahbron/avr-hal#quickstart).  You should build inside one of the crate directories, eg. in your case `boards/arduino-uno/`.

jonahbron, 2019-06-26:
Yup, I think those errors were appearing because I build from the root folder.  It succeeds when I build from within the atmega32u4 chip dir.

closed 2019-06-26 by jonahbron

Rahix, 2019-06-27:
FYI, I made the error message in this case a little clearer.  It should now display

```text
error: You cannot select multiple chips at once!
  --> /home/rahix/.cargo/git/checkouts/avr-device-86e4c21e2d4dc300/f77864d/src/lib.rs:66:1
   |
66 | compile_error!("You cannot select multiple chips at once!");
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

error: Could not compile `avr-device`.
```

I know this still does not point to the root cause but at least it is better than the obscure error you got ... For a real fix we have to choose a design as described in Rahix/avr-device#17.