Rahix/avr-hal #6: Runtime?

jonahbron, 2019-06-25:
All of the chip implementations in the official Rust Embedded project are split into two packages: low level registry access and run-time.

Does that (or will it) map onto the two packages here (avr-device and avr-hal)?  Does (or will) avr-hal provide a run-time?

Rahix, 2019-06-25:
There is very little runtime code necessary for AVR, so I intend to have all of it in `avr-device`.

jonahbron, 2019-06-25:
> I intend to

That implies it's not done yet, yet the README has instructions on how to build a hex file ready to flash onto a board.  I totally don't understand how this stuff works, but it seems like a program can't work yet without a runtime?  Or am I missing something?

Rahix, 2019-06-25:
The bare minimum is there already, it is just not pretty yet.  For example, the main function has to be a

```rust
#[no_mangle]
pub extern fn main() -> ! {
    // ...
}
```

right now, in contrast to

```rust
#[entry]
fn main() -> ! {
    // ...
}
```

jonahbron, 2019-06-25:
@Rahix Ahhh I see.  Where in the code can I look to see how that works?   I've dug around but found nothing seems related to that.

Rahix, 2019-06-25:
The current "runtime" is generated by `avr-gcc`, so there is not much in any of my crates right now.  In the future, there should be a procedural macro `entry` as shown above that takes care of doing all that setup.  I am not yet sure how exactly that should look.

jonahbron, 2019-06-25:
Got it.  Thanks very much for explaining that.

closed 2019-06-25 by jonahbron