Rahix/avr-hal #14: How to get a pin's mask or pin number

Altaflux, 2019-07-07:
I cannot seem to figure out how can I retrieve the pin number programmatically.
My intention is do be able to write something similar to what you do in  /avr-generic/port.rs does but in my main code base:

```
unsafe {
    (*<$PORTX>::ptr()).$reg_port.modify(|r, w| {
        w.bits(r.bits() & !(1 << $i))
    });
}
```
For which I need $i.
But I am not able to find an way on how to obtain that number. Could you provide an example on how to achieve this?

Rahix, 2019-07-07:
I don't quite understand what you are trying to do.  Can you please clarify, what your goal is?

Altaflux, 2019-07-07:
Similar to what your macro I described above is doing. I want to change a bit on a register.
The bit on the register I want to change is the pin number:
Example from src/atmega328p/port.rs
`pc3: (PC3, 3)` The 3 here.

As currently analog read is not implemented I am trying to do it myself in the meantime:
`dp.ADC.admux.modify(|r, w| { w.bits(r.bits() | mask) }); //Enable ADC for a given pin`
But I need to know the mask of the pin. 
So, for a given pin, i need the mask to use.

Sorry if I am not making myself clear enough, let me know if more info is needed and thanks for the help and effort!

Rahix, 2019-07-08:
Still not quite sure ... But I think I got something for you:

If you downgrade the pins, you can match them like this (or slightly different, this code is untested):

```rust
let pin = pc3.downgrade();

if let atmega328p_hal::port::Pin::C(id, _) = pin {
    // ...
} else {
    panic!("Only pins of port C can be used for analog reads!");
}
```

Altaflux, 2019-07-08:
Yeah! This is exactly what I was looking for, that "id". Only correction "i think" is that this is the correct import that worked for me:
```
let pin = pc3.downgrade();

if let arduino_uno::hal::port::Pin::C(id, _) = pin {
    // ...
} else {
    panic!("Only pins of port C can be used for analog reads!");
}
```
Thanks a lot for the help and assistance! Great work!

closed 2019-07-08 by Altaflux