Rahix/avr-hal #15: Unsafe macros break build

jonahbron, 2019-10-23:
Looks like the commit that went in yesterday (0aef1a6) fixed one layer of the issue, but there seem to be remaining issues.  Here's what I got when trying to build today:

```
error[E0133]: call to unsafe function is unsafe and requires unsafe function or block
  --> /home/jonah/Projects/avr-hal/chips/atmega328p-hal/src/spi.rs:31:1
   |
31 | / avr_hal::impl_spi! {
32 | |     pub struct Spi {
33 | |         peripheral: crate::atmega328p::SPI,
34 | |         pins: {
...  |
39 | |     }
40 | | }
   | |_^ call to unsafe function
   |
   = note: consult the function's documentation for information on how to avoid undefined behavior
   = note: this error originates in a macro outside of the current crate (in Nightly builds, run with -Z external-macro-backtrace for more info)
```

Rahix, 2019-10-24:
I think you have already fixed this in commit bd13c70417da ("Wrapped newly unsafe function in an unsafe block")?

jonahbron, 2020-02-06:
Yeah I think so, I've had no trouble compiling recently.

closed 2020-02-06 by jonahbron