Rahix/avr-hal #16: Open drain behaviour simulation

rumatoest, 2019-10-24:
Hello.

As I can see it is not possible to have Pin that implements InputPin + OutputPin trait at the same time.
Plus because on changing pin mode it consumes itself implementing such behavior is extremely hard in 3rd party code.

But the thing is that Arduino have modules that  communicates via single wire with pull up resistor in both direction (send/receive data). From Arduino libraries you can see that they try to emulate open drain behavior by switching pin in input mode when required to set high state and read it and only to set_low logic pin switched into output mode.

It would be really nice to have such behavior here with  InputPin + OutputPin traits implementation at the same time.

Rahix, 2020-07-28:
I think it should be possible to add an additional [mode](https://github.com/Rahix/avr-hal/blob/master/avr-hal-generic/src/port.rs#L4) where the pin type does exactly what you describe.  Though I'm not sure what the semantics should be when e.g. setting the pin low and _then_ reading.  This would 'toggle' the state implicitly and this could be confusing ...

jonahbron, 2020-07-28:
What about another pin struct that implements both traits, that can internally switch itself?  Something roughly like this (likely full of mistakes)
```
enum InputOrOutputPin {
    Output(OutputPin),
    Input(InputPin),
}
struct InputOutputPin {
    pin: InputOrOutputPin
}

impl InputOutputPin {
    def get_output(&mut self) -> OutputPin {
        match self.pin {
            Output(output_pin) => {
                output_pin
            },
            Input(input_pin) => {
                self.pin = Output(input_pin.release().into_output())
            },
        }
    }
    def get_input(&mut self) -> InputPin {
        match self.pin {
            Input(input_pin) => {
                input_pin
            },
            Output(output_pin) => {
                self.pin = Input(output.release().into_input())
            },
        }
    }
}

impl OutputPin for InputOutputPin {
    def set_high(&mut self) -> () {
        self.get_output().set_high()
    }
    def set_low(&mut self) -> () {
        self.get_output().set_low()
    }
}
impl InputPin for InputOutputPin {
    def is_high(&mut self) -> () {
        self.get_input().is_high()
    }
    def is_low(&mut self) -> () {
        self.get_input().is_low()
    }
}
```

kallemooo, 2020-08-10:
The proposed InputOutputPin will not work with the one-wire-bus library as it expects the set_high() to release the bus.
The InputOutputPin approach will force the the pin to high state (PushPull).
In open drain mode you shall never force the pin high (using push-pull) as other devices on the bus can set the bus low at the same time, and that will cause a high current rush, and possible break the devices.

Emulating an open drain pin on the AVR in a glitch safe mode needs to switch between input and output
mode as part of the emulation. To do that I see the following approach.

- **Init state**: Input floating. (DDRxn = 0, PORTxn = 0) 
- **set_low()**: Switch to output low mode (DDRxn = 1). This will force the pin low.
- **set_high()**:Switch to input floating mode. (DDRxn = 0). The external pull-up will set the pin high.
. **is_high() and is_low()**: Read the PINxn register. Do not switch mode.

I would suggest to follow the STM32-hal approach with modes for output `Output<OpenDrain>` and `Output<PushPull>`.
But that will break the current API for Output pin, but I do not know if it is a problem.

### Note:
Do not use the internal pull-up when emulating open drain mode.
As you cannot switch from input pull-up to output low in one operation.
It always needs two writes. And that can cause glitches on the bus, or high current rush depending on the approach.

Rahix, 2020-08-10:
> Emulating an open drain pin

Is it still 'emulating' at this point?  If I understand correctly, this _is_ electrically what an open-drain pin does, no?

> I would suggest to follow the STM32-hal approach with modes for output Output<OpenDrain> and Output<PushPull>.

I would add a `mode::OpenDrainIO` (or just `mode::OpenDrain`, not sure what makes more sense) where pins implement both input and output traits.

kallemooo, 2020-08-10:
> > Emulating an open drain pin
> 
> Is it still 'emulating' at this point? If I understand correctly, this _is_ electrically what an open-drain pin does, no?
It _is_ electrically what an open-drain pin does, but as the HW do not directly support the mode it is sort of a SW emulation.
But on electrical level is not an emulation.

> > I would suggest to follow the STM32-hal approach with modes for output Output and Output.
> 
> I would add a `mode::OpenDrainIO` (or just `mode::OpenDrain`, not sure what makes more sense) where pins implement both input and output traits (as you described, using DDR for switching the state).
I looked at the code and I think a new mode `mode::OpenDrain` is the best approach.

Rahix, 2020-08-10:
@kallemooo, would you be interested in implementing this?  Or would you rather have someone else take this upon themselves? ;)

kallemooo, 2020-08-10:
I need it for my temperature sensor project, so I will give it a try.

Rahix, 2020-08-10:
That would be awesome!  If you need any help, let me know; either here or in the `#embedded-rust` channel!

couchand, 2020-08-17:
A brief note on the terminology: AVRs all have totem pole output stages, so the term "open drain" is strictly incorrect.  I think tri-state output would be the usual term in this case.

rumatoest, 2020-08-17:
I'm not good in terms. But the main thing is that such pin could only do next things:

1. Read input when we need to read pin
2. If we need to set it to LOW state -> switch mode to output and do -> set low thing
3. If we need to set "HIGHT" state -> just switch to input mode (considering that there is already pull up resistor in circuit)

Just look into DHT drivers from Adafruit - they do similar stuff in C++ code.

commit e00abb6c (2020-08-18): generic: port: Support for open-drain mode

A new port mode (OpenDrain) that suports open-drain mode with external pull-up.
The solution uses the DDRxn to switch between a high state with the pin
input floating mode where the external pull-up is setting the pin to high state,
and low state by setting the pin in output low to pull the pin low.

Input traits is reading the external electrical state independent of if
the pin is set high or low.

Closes #16.
closed 2020-08-18 by Rahix