Rahix/avr-hal #19: consider migrating to the new asm! syntax specified in RFC 2873

jdrouet, 2020-07-14:
Hi!
I've been trying to following code example for `atmega328p`:

```rust
#![no_std]
#![no_main]
#![feature(proc_macro_hygiene)]

use arduino_leonardo::prelude::*;

#[panic_handler]
fn panic(info: &core::panic::PanicInfo) -> ! {
    let mut serial: arduino_leonardo::Serial<arduino_leonardo::hal::port::mode::Floating> = unsafe {
        core::mem::uninitialized()
    };

    ufmt::uwriteln!(&mut serial, "Firmware panic!\r");

    if let Some(loc) = info.location() {
        ufmt::uwriteln!(
            &mut serial,
            "  At {}:{}:{}\r",
            loc.file(),
            loc.line(),
            loc.column(),
        );
    }

    loop {}
}

#[no_mangle]
pub extern fn main() -> ! {
    let dp = arduino_leonardo::Peripherals::take().unwrap();

    let mut pins = arduino_leonardo::Pins::new(
        dp.PORTB,
        dp.PORTC,
        dp.PORTD,
        dp.PORTE,
    );

    let mut serial = arduino_leonardo::Serial::new(
        dp.USART1,
        pins.d0,
        pins.d1.into_output(&mut pins.ddr),
        57600,
    );

    ufmt::uwriteln!(&mut serial, "Hello from Arduino!\r").unwrap();
    // Panic messages cannot yet be captured because they rely on core::fmt
    // which is way too big for AVR
    panic!();
}
```

My `Cargo.toml` is the following

```rust
[package]
name = "sensor"
version = "0.1.0"
authors = ["Jérémie Drouet <jeremie.drouet@gmail.com>"]
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
panic-halt = "0.2.0"

[dependencies.arduino-leonardo]
path = ".../avr-hal/boards/arduino-leonardo"

[profile.dev]
panic = "abort"
codegen-units = 1
incremental = false
lto = true

[profile.release]
panic = "abort"
codegen-units = 1
debug = false
lto = true
```

I'm using rust sources from [this PR](https://github.com/rust-lang/rust/pull/73270) and [the blink example](https://github.com/avr-rust/blink) compiles well.

But when I try to use `avr-hal`, I get the following error.

```
error: the legacy LLVM-style asm! syntax is no longer supported
  --> /home/jdrouet/.cargo/git/checkouts/avr-device-86e4c21e2d4dc300/67d2554/src/interrupt.rs:17:9
   |
17 |           asm!(
   |           ^---
   |           |
   |  _________help: replace with: `llvm_asm!`
   | |
18 | |             "cli" :::: "volatile"
19 | |         );
   | |__________^
   |
   = note: consider migrating to the new asm! syntax specified in RFC 2873
   = note: alternatively, switch to llvm_asm! to keep your code working as it is

error: the legacy LLVM-style asm! syntax is no longer supported
  --> /home/jdrouet/.cargo/git/checkouts/avr-device-86e4c21e2d4dc300/67d2554/src/interrupt.rs:31:9
   |
31 |           asm!(
   |           ^---
   |           |
   |  _________help: replace with: `llvm_asm!`
   | |
32 | |             "sei" :::: "volatile"
33 | |         );
   | |__________^
   |
   = note: consider migrating to the new asm! syntax specified in RFC 2873
   = note: alternatively, switch to llvm_asm! to keep your code working as it is

error: the legacy LLVM-style asm! syntax is no longer supported
  --> /home/jdrouet/.cargo/git/checkouts/avr-device-86e4c21e2d4dc300/67d2554/src/interrupt.rs:48:9
   |
48 |           asm!(
   |           ^---
   |           |
   |  _________help: replace with: `llvm_asm!`
   | |
49 | |             "in $0,0x35"
50 | |             : "=r"(sreg)
51 | |             :
52 | |             :
53 | |             : "volatile"
54 | |         );
   | |__________^
   |
   = note: consider migrating to the new asm! syntax specified in RFC 2873
   = note: alternatively, switch to llvm_asm! to keep your code working as it is

error: aborting due to 3 previous errors

error: could not compile `avr-device`
```

Do you have any idea on how I could fix that?

Rahix, 2020-07-14:
Yeah, I'm aware of this change and have actually already prepared a patch locally.  I just held off from pushing it here because I am waiting for upstream nightly rust to be able to compile libcore.  Once that works (looks like the PR you mentioned will archieve this?), I'll update!

For the time being, you can apply the patch locally: [0001-Switch-to-llvm_asm-to-fix-build-on-latest-nightly.patch.zip](https://github.com/Rahix/avr-hal/files/4920149/0001-Switch-to-llvm_asm-to-fix-build-on-latest-nightly.patch.zip) (zip-archived because github doesn't let me upload `.patch` files ...)

---

A different topic: The files you showed reference `arduino-leonardo` which uses an `ATmega32U4`, not an `ATmega328P`.  So you'll probably want to switch to the `arduino-uno` crate instead ...

Rahix, 2020-07-14:
(Forgot to mention: This patch goes ontop of [`avr-device`](https://github.com/Rahix/avr-device) of course, not `avr-hal`)

lperlaki, 2020-07-24:
Since [rust/73270](https://github.com/rust-lang/rust/pull/73270) got merged I open a pull request #20

Rahix, 2020-07-24:
Merged!  I suppose I should also update the build-instructions now as the
custom toolchain is no longer needed which greatly simplifies things ...

closed 2020-07-24 by Rahix