Rahix/avr-hal #21: Remove "singlethread": true

lights0123, 2020-07-24:
In all the target JSON files, `"singlethread": true` is set. However, I don't believe that this works—[interrupts can still cause problems](https://github.com/rust-lang/rust/issues/58500#issuecomment-654341233), as an interrupt happening at the wrong place can interfere with the operation. The only operation that AVR can perform safely is an 8-bit read or write. All other operations require disabling interrupts.

Rahix, 2020-07-24:
Right, I wasn't aware of the true meaning of the field.  Thanks for pointing it out to me!

closed 2020-07-24 by Rahix

lights0123, 2020-07-24:
For anyone coming across this in the future, I wrote a guide about what to do if you want to use the Atomic types: https://docs.rust-embedded.org/embedonomicon/custom-target.html#fill-the-target-file

commit 1c74cfca (2020-07-24): Set "singlethread" to false in all target specs

'singlethread' should **only** be `true` for targets which have no form
of threading, interrupts, or similar.  This is clearly not true for AVR
so set 'singlethread' to false.

Fixes #21.