Rahix/avr-hal #29: Inline asm errors for uno-blink

Gregoor, 2020-07-29:
Heya,

just checked this out on Windows 10 with rust 1.47.0-nightly and tried to run the uno-blink example but got:

```
$ cargo build --example uno-blink
   Compiling arduino-uno v0.1.0 (C:\Users\grego\code\avr-hal\boards\arduino-uno)
error: unknown use of instruction mnemonic without a size suffix
  |
note: instantiated into assembly here
 --> <inline asm>:1:2
  |
1 |     in %al,0x35
  |     ^

error: invalid instruction mnemonic 'sei'
  |
note: instantiated into assembly here
 --> <inline asm>:1:2
  |
1 |     sei
  |     ^^^

error: aborting due to 2 previous errors

error: could not compile `arduino-uno`.
```

Same problem in WSL 2 Ubuntu with the same rust version. Any pointers would be appreciated, maybe it's something I could look at myself.

lights0123, 2020-07-29:
`rustc --version`? I know that nightly-2020-07-27 works on Linux at least, and you need at least nightly-2020-07-24 (or maybe -23?) for it to build. I'm not running any nightly later than nightly-2020-07-27 yet, as rustfmt [isn't available for it](https://rust-lang.github.io/rustup-components-history/index.html).

Gregoor, 2020-07-29:
Huh! I was running on 2020-07-25, a `rustup update nightly` gave me 2020-07-26 which didn't fix it yet. I guess I'll try another update tomorrow or will install it without `rustup`!

lights0123, 2020-07-29:
And you've pulled this repository since [this commit 5 days ago](https://github.com/Rahix/avr-hal/commit/3cd813ec1df521615cc33408725b6bc338b84320), right?

Gregoor, 2020-07-29:
Yup, I'm on c8bde5f4b33835fe85160828a334088532876184

Rahix, 2020-07-29:
Huh, never seen that one before ... Looks strange to say the least. On first glance I'd say you are compiling for the wrong target but that doesn't seem to be the case ... Things to come to mind:

- Paste the log from `cargo build --verbose --examples`, maybe there's a hint hidden somewhere ...
- You can check whether `.cargo/config.toml` in `boards/arduino-uno` exists and has the contents shown here on GitHub.
- Do you have a proper version of `avr-gcc`, `avr-binutils`, and `avr-libc` installed?  Can you also show what versions?
- Run `cargo update` in case you are pulling in an old dependency in some weird way.
  - Or go nuclear and remove any `Cargo.lock` + `target/` directory + possibly `sccache` directory (if you're using that).

Gregoor, 2020-07-29:
I'm not sure if I have `avr-binutils` installed but the rest I've got from WINAVR which is.. _adjusts glasses_ ... from 2010:
```bash
$ avr-gcc  --version
avr-gcc.exe (WinAVR 20100110) 4.3.3

$ avr-cpp.exe --version
avr-cpp.exe (WinAVR 20100110) 4.3.3
```

<details>
  <summary>Here's the cargo build</summary>

```
$ cargo build --verbose --examples
       Fresh unicode-xid v0.2.1
       Fresh semver-parser v0.7.0
       Fresh proc-macro-hack v0.5.18
       Fresh nb v1.0.0
       Fresh void v1.0.2
       Fresh ufmt-write v0.1.0 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)       Fresh cfg-if v0.1.10
       Fresh vcell v0.1.2
       Fresh panic-halt v0.2.0
       Fresh semver v0.9.0
       Fresh nb v0.1.3
       Fresh paste-impl v0.1.18
       Fresh proc-macro2 v1.0.19
       Fresh rustc_version v0.2.3
       Fresh embedded-hal v0.2.4
       Fresh quote v1.0.7
       Fresh paste v0.1.18
       Fresh syn v1.0.36
       Fresh bare-metal v0.2.5
       Fresh ufmt-macros v0.1.1 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
       Fresh avr-device-macros v0.1.0
       Fresh ufmt v0.1.0 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
       Fresh avr-device v0.1.0
       Fresh avr-hal-generic v0.1.0 (C:\Users\grego\code\avr-hal\avr-hal-generic)
       Fresh atmega32u4-hal v0.1.0 (C:\Users\grego\code\avr-hal\chips\atmega32u4-hal)
       Fresh atmega328p-hal v0.1.0 (C:\Users\grego\code\avr-hal\chips\atmega328p-hal)
       Fresh atmega1280-hal v0.1.0 (C:\Users\grego\code\avr-hal\chips\atmega1280-hal)
       Fresh attiny85-hal v0.1.0 (C:\Users\grego\code\avr-hal\chips\attiny85-hal)
   Compiling arduino-leonardo v0.1.0 (C:\Users\grego\code\avr-hal\boards\arduino-leonardo)
   Compiling bigavr6 v0.1.0 (C:\Users\grego\code\avr-hal\boards\bigavr6)
     Running `rustc --crate-name arduino_leonardo --edition=2018 'boards\arduino-leonardo\src\lib.rs' --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -Clinker-plugin-lto -C debuginfo=2 -C debug-assertions=on --cfg 'feature="default"' --cfg 'feature="rt"' -C metadata=743d4ac3215ce35d -C extra-filename=-743d4ac3215ce35d --out-dir 'C:\Users\grego\code\avr-hal\target\debug\deps' -C 'incremental=C:\Users\grego\code\avr-hal\target\debug\incremental' -L 'dependency=C:\Users\grego\code\avr-hal\target\debug\deps' --extern 'atmega32u4_hal=C:\Users\grego\code\avr-hal\target\debug\deps\libatmega32u4_hal-05ec469b8608b567.rmeta' --extern 'avr_hal_generic=C:\Users\grego\code\avr-hal\target\debug\deps\libavr_hal_generic-3e975191fba3f363.rmeta'`
   Compiling trinket v0.1.0 (C:\Users\grego\code\avr-hal\boards\trinket)
   Compiling arduino-uno v0.1.0 (C:\Users\grego\code\avr-hal\boards\arduino-uno)
     Running `rustc --crate-name bigavr6 --edition=2018 'boards\bigavr6\src\lib.rs' --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -Clinker-plugin-lto -C debuginfo=2 -C debug-assertions=on --cfg 'feature="default"' --cfg 'feature="rt"' -C metadata=eb7bd84a0ba7a8a1 -C extra-filename=-eb7bd84a0ba7a8a1 --out-dir 'C:\Users\grego\code\avr-hal\target\debug\deps' -C 'incremental=C:\Users\grego\code\avr-hal\target\debug\incremental' -L 'dependency=C:\Users\grego\code\avr-hal\target\debug\deps' --extern 'atmega1280_hal=C:\Users\grego\code\avr-hal\target\debug\deps\libatmega1280_hal-f37c77eb46e01ad6.rmeta' --extern 'avr_hal_generic=C:\Users\grego\code\avr-hal\target\debug\deps\libavr_hal_generic-3e975191fba3f363.rmeta'`
     Running `rustc --crate-name trinket --edition=2018 'boards\trinket\src\lib.rs' --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -Clinker-plugin-lto -C debuginfo=2 -C debug-assertions=on --cfg 'feature="default"' --cfg 'feature="rt"' -C metadata=c5b5d45392ab0113 -C extra-filename=-c5b5d45392ab0113 --out-dir 'C:\Users\grego\code\avr-hal\target\debug\deps' -C 'incremental=C:\Users\grego\code\avr-hal\target\debug\incremental' -L 'dependency=C:\Users\grego\code\avr-hal\target\debug\deps' --extern 'attiny85_hal=C:\Users\grego\code\avr-hal\target\debug\deps\libattiny85_hal-55c9383553d9b936.rmeta' --extern 'avr_hal_generic=C:\Users\grego\code\avr-hal\target\debug\deps\libavr_hal_generic-3e975191fba3f363.rmeta'`
     Running `rustc --crate-name uno_serial --edition=2018 'boards\arduino-uno\examples\uno-serial.rs' --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C opt-level=s -C panic=abort -C lto -C debuginfo=2 -C debug-assertions=on --cfg 'feature="default"' --cfg 'feature="rt"' -C metadata=07d5d3dddbf556c2 --out-dir 'C:\Users\grego\code\avr-hal\target\debug\examples' -C 'incremental=C:\Users\grego\code\avr-hal\target\debug\incremental' -L 'dependency=C:\Users\grego\code\avr-hal\target\debug\deps' --extern 'arduino_uno=C:\Users\grego\code\avr-hal\target\debug\deps\libarduino_uno-61bbfb7e45f58e23.rlib' --extern 'atmega328p_hal=C:\Users\grego\code\avr-hal\target\debug\deps\libatmega328p_hal-31882f687be1c3d2.rlib' --extern 'avr_hal_generic=C:\Users\grego\code\avr-hal\target\debug\deps\libavr_hal_generic-3e975191fba3f363.rlib' --extern 'nb=C:\Users\grego\code\avr-hal\target\debug\deps\libnb-b1a0beafcc543e3f.rlib' --extern 'panic_halt=C:\Users\grego\code\avr-hal\target\debug\deps\libpanic_ha [... cut ...]
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target '\\?\C:\Users\grego\code\avr-hal\boards\arduino-uno\avr-atmega328p.json' --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit code: 1)
  --- stderr
  error: Error loading target specification: SyntaxError(InvalidSyntax, 1, 1). Use `--print target-list` for a list of built-in targets
```
Can't say that I learned anything from that error yet. My .cargo/config.toml is the same as in master indeed.

lights0123, 2020-07-29:
Ok, it looks like Rust isn't picking up on the target. I wonder if that's because NTFS doesn't like the fact that `
avr-atmega328p.json` in boards/arduino-uno is a symlink? Try replacing the file with boards/arduino-uno/avr-atmega328p.json with chips/atmega328p-hal/avr-atmega328p.json (just copy it).

Gregoor, 2020-07-29:
Oh I don't know how you got that from that but it seemed to have done the trick. Thank you for looking into it with me. Anything I can do to prevent other NTFS-people from running into it?

lights0123, 2020-07-29:
I figured that out because the 6th line in you last code block indicated that it had trouble reading the file, and seeing the path `\\?\C:\Users\grego\code\avr-hal\boards\arduino-uno\avr-atmega328p.json` made me think that maybe some weird Windows FS thing was going on.

It turns out that Windows *still* doesn't support symlinks by non-admin users by default, wtf! To fix this ([stolen from StackOverflow](https://stackoverflow.com/a/59761201/4471524)):

- Enable "Developer mode" in Windows 10
- `git config --global core.symlinks true`
- re-clone this repo, or switch to another branch and back (`git checkout -b temp && git checkout master`)

@Rahix, we should either document this or remove symlinks.

Rahix, 2020-07-29:
> It turns out that Windows still doesn't support symlinks by non-admin users by default, wtf!

What a mess ...

> @Rahix, we should either document this or remove symlinks.

Right, I'll think about it.  Now that we can use `.cargo/config.toml` for naming the target json files, I think it would be okay to do without the symlinks.  Previously this would mean additional work because you'd have to `export RUSTC_TARGET_PATH` and I wanted it easy with `export RUSTC_TARGET_PATH=$(pwd)` instead of `export RUSTC_TARGET_PATH=/path/to/wherevever/the/actual/target/file/is/`.

Gregoor, 2020-07-29:
> It turns out that Windows _still_ doesn't support symlinks by non-admin users by default, wtf! To fix this ([stolen from StackOverflow](https://stackoverflow.com/a/59761201/4471524)):
>
> * Enable [...]

Yup that worked too! Though a branch switch wasn't enough for me, had to re-clone. Thanks again :)

Rahix, 2020-10-07:
No more symlinks in this repository so hopefully no more build-issues on windows!

commit 539924f1 (2020-10-07): treewide: Move all AVR target specs into a single directory

This allows us to remove the symlinks and later add
infrastructure for automatically deriving these files
from the upstream target spec.

Fixes #29 
Ref #78
closed 2020-10-07 by Rahix