Rahix/avr-hal #32: Cleanup target files

Rahix, 2020-08-03:
Notes to myself about what needs to happen:

- [ ] Move all target files into a single top-level directory.  Update all `.cargo/config.toml`s to point to the files in this place instead.  ->  This gets rid of the symlinks thus fixing #29.
- [ ] Do [this cleanup](https://github.com/avr-rust/blink/pull/20) to all our target files as well.
- [ ] Check remaining settings against current upstream defaults (after https://github.com/rust-lang/rust/pull/74941)
- [ ] Think about maybe adding infrastructure for automatically deriving the target files from upstream defaults to reduce maintenance burden.

explicite, 2020-10-03:
@Rahix  can you please mark second checkbox?

explicite, 2020-10-04:
Looks line in third one we have some additional setting which I cannot add find in our targets. 
https://github.com/rust-lang/rust/pull/74941/files#diff-e26bdcc4844624b6b557a83e64e20c0a
I cannot see  e.g `eh_frame_header` or `max-atomic-width`. Should we support all old and new options or remove legacy from our targets? Please find bellow all merged.
```
{
  "arch": "avr",
  "data-layout": "e-P1-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8",
  "llvm-target": "avr-unknown-unknown",
  "target-endian": "little",
  "target-pointer-width": "16",
  "linker-flavor": "gcc",
  "os": "unknown",
  "target-env": "",
  "target-vendor": "unknown",
  "target-c-int-width": "16",
  "cpu": "atmega2560",
  "exe-suffix": ".elf",
  "linker": "avr-gcc",
  "dynamic_linking": false,
  "executables": true,
  "linker_is_gnu": true,
  "has_rpath": false,
  "position_independent_executables": false,
  "eh_frame_header": false,
  "max-atomic-width": 8,
  "pre-link-args": {
    "gcc": [
      "-mmcu=atmega2560"
    ]
  },
  
  "post-link-args": {
    "gcc": [
      "-Wl,--gc-sections"
    ]
  },
  "singlethread": false,
  "no-builtins": false,
  "no-default-libraries": false,
  "eh-frame-header": false
}
```

Rahix, 2020-10-04:
This command shows the current upstream default target:

```bash
rustc --print target-spec-json -Z unstable-options --target avr-unknown-gnu-atmega328
```

Rahix, 2020-10-04:
@explicite, I suggest not spending too much time on this now because as I wrote in the last bullet point, ideally I want to move towards autogeneration of these target files.

explicite, 2020-10-05:
@Rahix, maybe I can use https://github.com/serde-rs/json and create some trait with default impl for generic. Then I will override per chip? Then `magic` can be done in `build.rs`?

Rahix, 2020-10-06:
@explicite, no the target file must exist before invoking cargo so this can't happen in `build.rs`.  I do want to keep the target files checked into the repo for that reason.  My idea was this:

- A toplevel directory named `avr-specs/` (bikeshedding welcome) containing all the target specs we have.
- There's a small python script named `avr-specs/sync-from-upstream.py` which runs rustc to retrieve the current upstream spec, patches the few things necessary, and overwrites the files.
- I'll then run this script from time to time to keep our specs up-to-date.

PR #85 (2020-10-09): Bring target specs in line with upstream
@explicite, I've reviewed all the changes which upstream did since I initially created these target files.  I have carefully done each change as a separate commit so it is possible to retrace what happened.  As far as I can tell, this does not produce any regressions in avr-hal, but let's see what CI has to say about this.

Finally fixes #32.
closed 2020-10-09 by Rahix