Rahix/avr-hal #37: error: Could not create LLVM TargetMachine for triple: avr-unknown-unknown: No available targets are compatible with triple "avr-unknown-unknown"

Dajamante, 2020-08-06:
Hi,
I am working from the avr-hal basic template project.

When I try to refresh the cargo project(I am on intelliJ ultimate) project I get error:
[](url)
10:24	Cargo project update failed:
			Execution failed (exit code 101).
			        /Users/aissata/.cargo/bin/cargo metadata --verbose --format-version 1 --all-features
			        stdout : error: failed to run `rustc` to learn about target-specific information
			
			Caused by:
			  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target /Users/aissata/Rust/avr-example/avr-atmega328p.json --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit code: 1)
			--- stderr
error: Could not create LLVM TargetMachine for triple: avr-unknown-unknown: No available targets are compatible with triple "avr-unknown-unknown"

Rahix, 2020-08-06:
Okay, can you check a) what version of rust nightly you have installed and b) whether intelliJ is actually using rust nightly?  The error you are seeing indicates that whatever rust compiler was invoked by intelliJ does not support AVR.  This could mean you are accidentally using rust stable or you have a version of rust nightly which is too old.

Dajamante, 2020-08-06:
I did rustup:
➜ rustup show
Default host: x86_64-apple-darwin
rustup home:  /Users/aissata/.rustup

installed toolchains
--------------------

stable-x86_64-apple-darwin
nightly-x86_64-apple-darwin

active toolchain
----------------

stable-x86_64-apple-darwin (default)
rustc 1.45.2 (d3fb005a3 2020-07-31)

It's up to date. Tried to override to nightly to make sure:

➜ rustup override set nightly
info: using existing install for 'nightly-x86_64-apple-darwin'
info: override toolchain for '/Users/aissata/Rust/avr-example' set to 'nightly-x86_64-apple-darwin'

  nightly-x86_64-apple-darwin unchanged - rustc 1.47.0-nightly (6c8927b0c 2020-07-26)

In intelliJ in debug configurations I made the Channel to nightly.

I still get the error.

Rahix, 2020-08-06:
Nice!  Maybe we should somehow mention this in the step-by-step guide?

Dajamante, 2020-08-06:
My plan is to do a step-by-step for the utterly beginner, when I managed to make the car robot work (or slightly before maybe) 👍. Maybe when my sensor is working and printing on screen I can make a PR with an example?

closed 2020-08-06 by Dajamante