Rahix/avr-hal #39: Assembly issues

foosinn, 2020-08-08:
This seems to be related to #29, but happens on linux, even after replacing all symlinks

Steps the reproduce:

```sh
sudo docker run -it debian:buster-slim
apt update && apt install -y curl git gcc-avr build-essential
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh  # select nightly 
. /root/.profile
git clone https://github.com/Rahix/avr-hal.git
cd acr-hal
cargo build --examples
```
This leads to the following error:

```
root@f54a89809b4a:/avr-hal# cargo build --examples
   Compiling proc-macro2 v1.0.19
   Compiling syn v1.0.38
   Compiling proc-macro-hack v0.5.18
   Compiling nb v0.1.3
   Compiling rustc_version v0.2.3
   Compiling embedded-hal v0.2.4
   Compiling bare-metal v0.2.5
   Compiling quote v1.0.7
   Compiling paste-impl v0.1.18
   Compiling paste v0.1.18
   Compiling avr-device-macros v0.2.0
   Compiling ufmt-macros v0.1.1 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
   Compiling avr-device v0.2.1
   Compiling ufmt v0.1.0 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
   Compiling avr-hal-generic v0.1.0 (/avr-hal/avr-hal-generic)
   Compiling atmega32u4-hal v0.1.0 (/avr-hal/chips/atmega32u4-hal)
   Compiling atmega328p-hal v0.1.0 (/avr-hal/chips/atmega328p-hal)
   Compiling atmega2560-hal v0.1.0 (/avr-hal/chips/atmega2560-hal)
   Compiling atmega1280-hal v0.1.0 (/avr-hal/chips/atmega1280-hal)
   Compiling attiny85-hal v0.1.0 (/avr-hal/chips/attiny85-hal)
   Compiling arduino-uno v0.1.0 (/avr-hal/boards/arduino-uno)
   Compiling arduino-leonardo v0.1.0 (/avr-hal/boards/arduino-leonardo)
   Compiling trinket v0.1.0 (/avr-hal/boards/trinket)
error: unknown use of instruction mnemonic without a size suffix
  |
note: instantiated into assembly here
 --> <inline asm>:1:2
  |
1 |     in %al,0x3F
  |     ^

error: invalid instruction mnemonic 'sei'
  |
note: instantiated into assembly here
 --> <inline asm>:1:2
  |
1 |     sei
  |     ^^^

error: aborting due to 2 previous errors

error: could not compile `arduino-uno`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: unknown use of instruction mnemonic without a size suffix
  |
note: instantiated into assembly here
 --> <inline asm>:1:2
  |
1 |     in %al,0x3F
  |     ^

error: invalid instruction mnemonic 'sei'
  |
note: instantiated into assembly here
 --> <inline asm>:1:2
  |
1 |     sei
  |     ^^^

error: aborting due to 2 previous errors

error: build failed
```

Thanks a lot for your time and efforts, it's more than awesome to switch away from the arduino ide to rust!

foosinn, 2020-08-08:
I also was able to reproduce this error on Arch Linux and NixOS

Rahix, 2020-08-08:
I think you forgot to switch into a board directory.  You'll want to build e.g. from inside `boards/arduino-uno`, not from the top-level directory as each MCU has its own target configuration which are only present in those directories.

foosinn, 2020-08-10:
I did not setup `.cargo/config.toml` correctly. For anyone stumbling on this:

```
$ cat .cargo/config.toml
[build]
target = "avr-atmega32u4.json"

[unstable]
build-std = ["core"]
```

closed 2020-08-10 by foosinn