Rahix/avr-hal #42: Add ability to re-configure SPI.

davidlenfesty, 2020-08-09:
I'm trying to use SPI to read off an SD card, and part of the requirements of an SD card is that it needs to initialize at 100-400kHz, but then you can run it faster afterwards.

I'm using `embedded-sdmmc`, and it allows you to reconfigure the peripheral after initializing through a function. However as this HAL currently exists, there is no way to do that other than releasing the existing instance and creating a new one.

I have tested the following and it seems to work (haven't built the hardware to test yet):

``` rust
impl $Spi {
    [...]
    /// Reconfigure SPI peripheral without releasing pins.
    pub fn reconfigure(&mut self, settings: Settings) {
        self.settings = settings;
        self.setup();
    }
    [...]
}
```

Rahix, 2020-08-09:
cc @jonahbron, what do you think?

couchand, 2020-08-14:
Presumably you'd want to wait for any in-flight writes to complete, so perhaps the implementation would be:

```
pub fn reconfigure(&mut self, settings: Settings) -> nb::Result<(), Void> {
    self.flush()?;
    self.settings = settings;
    self.setup();
    Ok(())
}
```

closed 2020-12-07 by Rahix