Rahix/avr-hal #44: Accurate PWM

Rahix, 2020-08-10:
The existing PWM implementation is fixed to 1kHz and uses the _Fast PWM_ mode which isn't phase- nor frequency-correct.  This is rather useless for applications like controlling a Servo where an exact frequency and duty-cycle are needed.

So, in addition to the _Fast PWM_ implementation a seconds "correct" PWM implementation should be added which allows specifying the frequency + duty cycle exactly.  Ideally, this should be based on the _rate_ and _duration_ primitives from [`embedded-time`](https://docs.rs/embedded-time/0.9.1/embedded_time/index.html).

Open Questions:
- How to select the prescaling?  We could devise an algorithm to use the smallest clock-divider which still fits the frequency.  This would, however make it entirely unobvious what precision the duty-cycle will have.  Would good documentation help here?  Or should the prescaler be selected manually instead?
- Should the `Duty` type be a percentage or a _duration_?

explicite, 2020-11-24:
1. Looks like prescaler is better option. It's better to be explicite
1. [Duration](https://docs.rs/embedded-time/0.9.1/embedded_time/duration/trait.Duration.html)? Percentage seems not very precise. Percentage can help in some use cases?  
1. Are we able to define this on avr-device?

Rahix, 2020-11-25:
> Looks like prescaler is better option. It's better to be explicite

Not sure what you mean?  We'll use the prescaler either way, the question is how it is initialized.  Either we have an algorithm which takes a frequency and selects a fitting prescaler, or we leave it up to the user to decide manually.

Actually, while thinking about it: Why not both?  We could provide two separate constructors for 'automatic' and 'manual' configuration ...

> Duration? Percentage seems not very precise.

In terms of precision, the two solutions don't differ too much.  It's always a bit up to luck whether the integer values cleanly divide the timer steps.  But we could, again, build a solution where all possibilities are accounted for and provide multiple methods for setting the duty-cycle based on exact timer counts, based on duration, or based on percentage.

> Aret there any usecases where percentage is common?

Yes, definitely.  PWM is a cheap way to create power output that scales linearly and for those usecases you mostly care about a percentage of the maximum power (= 100% duty cycle).

> Are we able to define this in avr-device?

No, this belongs into the HAL only.

koutoftimer, 2021-02-11:
IMHO, hardware abstraction is creative process, it is a search of something new, something that didn't exist. I would like to stick with reality and make a handy interface, at least in the beginning.

Lets look at `Pins` from atmega328p. `avr-device` generates pretty neat interface for chip's internals. Arduino Uno, on the other hand, provides different pinout. In order to use Arduino Uno you have to translate it's own pins to MCU's pins. Interfacing `Pins` struct from this perspective is much more handier if you are working with Arduino Uno.

Suppose you want to generate some form of signal on specific pin of Arduino board. Intuitive way is to access needed pin to setup wave parameters, but right way is to find out what timer is needed first.

*First proposal:* add timer access via pin.

Why you need to access timer? Lets take at [SG 90][servo] datasheet. Wave period 20 ms with 1-2 ms duty cycle mapped for -90..90 rotation angle. Internal generator frequency 16 MHz, with prescaler 1024 maximum period we can get with 8 bit timer is ~16.3 ms (~81% of required time). Let's imagine that we are fine with this period, next thing we need is to map -90..90 angle range to 1-2 ms duty cycle range. 1 ms is about 16 cycles. 180 degrees / 16 = 11.5 degrees per step is how accurate value we can set. It is a huge difference in angle. Such steps are pretty fine for LEDs if you want smooth turn on\off, because you can not say for sure how much if differs. On the other hand, difference in 11 degrees everyone can see for sure. Situation is even worse if you need to map duty cycle for even shorter range or you want to handle sensitive PID controller.

Best thing you can do at this point is to use 16 bit register. `avr-device` does great job for interfacing all required registers to configure timers, but still fields like WGM, that are logically tighten, located in separate physical registers.

*Second proposal:* add data access layout to simplify field access of related `RegisterBlock`.

Timers have more use cases besides PWM and handy interface needed for sure. I have doubts that covering every possible use case with its own abstraction is good idea.

Unified 8 bit duty access can be achieved with default mapper that work similar to Arduino's `map` function by converting value from input range to value from output range. BTW, this is what I feel like best answer on duty type question.

*Third proposal:* add range mapper for PWM pins that will convert user defined type into timer's duty type. Default implementation should unify timer usage.

[servo]: http://www.ee.ic.ac.uk/pcheung/teaching/DE1_EE/stores/sg90_datasheet.pdf