Rahix/avr-hal #45: Support externally-pulled SPI

couchand, 2020-08-14:
The current implementation demands that `SCLK` and `MOSI` be in `mode::Output` and `MISO` be in `mode::Input<mode::PullUp>` before creating the `Spi`.  But the hardware would support more general bounds.

The input `MISO` could be floating, to allow for an external pull-up, and it seems this change can be made without any other code changes.

And while normal SPI operation is push-pull, it is occasionally configured as open-drain, which also seems to be supported by the hardware.  The datasheets clearly state the output pins' pin directions are user-defined.  Since I haven't found authoritative documentation on the matter and haven't set up a test bench, I'm reading between the lines a bit, but I interpret this to mean the SPI peripheral will output the signals by toggling the DDR rather than the pin.  Perhaps real testing is needed before making code changes here.

couchand, 2020-08-14:
On second thought, I don't think I want to go down the rabbit hole of figuring out what the user-defined direction on output pins means.  So I'll probably just open a PR making MISO input mode configurable and then close this issue.

Rahix, 2020-08-14:
For I2C I also made the pull-up configurable, maybe you can take that as a source of inspiration.

PR #49 (2020-08-19): Support floating MISO in SPI
Closes #45.
closed 2020-08-19 by Rahix