Rahix/avr-hal #53: More efficient pin toggling

Rahix, 2020-08-18:
Right now, pin-toggling is implemented naively using the default `embedded-hal` implementation:

https://github.com/Rahix/avr-hal/blob/f366645c7fc8260bde221975b74b38d8a824a1fe/avr-hal-generic/src/port.rs#L232-L234

This will essentially boil down to reading the current state, inverting it, and writing that back.  We can do better!  It turns out, by writing a `1` to the `PINn` register, the MCU will take care of toggling `PORTn`.  With the next version of `avr-device`, commit [7c3805f9d84d ("Make the PINx registers writeable")](https://github.com/Rahix/avr-device/commit/7c3805f9d84df7a1911ca3d29cfe28fa262c10b5) which was introduced by @couchand in PR Rahix/avr-device#45, this will actually be possible to implement here.

couchand, 2020-08-18:
I'm happy to take care of this if you aren't already on it, @Rahix.

Rahix, 2020-08-19:
Sure, that would be great!

PR #56 (2020-08-20): Single-instruction pin toggling.
Use the AVR-specific output pin toggle of writing to the PINx register,
rather than relying on the default implementation based on reading the
set state first.

Closes #53.
closed 2020-08-20 by Rahix