Rahix/avr-hal #54: Create a project template

Rahix, 2020-08-18:
Similar to [`cortex-m-quickstart`](https://github.com/rust-embedded/cortex-m-quickstart), we should provide a [`cargo-generate`](https://crates.io/crates/cargo-generate) template for AVR.  I'm not sure how exactly to design that to make it easy to start with any device from the same template but I'm sure there is a way to make this work.  Otherwise, several templates for different popular boards could also work.

ToddG, 2020-10-22:
I wrote this for my own use a few days ago: 

https://github.com/ToddG/arduino-rust-seed

Basically it uses python + cookiecutter to generate an application from templates.

koutoftimer, 2021-02-16:
I vote for orphan `template-{board name}` branches. Despite of branching support in liquid templates it will be a mess to write common template for every supported board. Separate branches are harder to support but easier to understand what is going on and easier to get involved for newcomers.

Rahix, 2021-02-17:
I like the idea of having the templates as branches in this repo!  We just need to make it super obvious where they are, because otherwise people will not find them...

I wouldn't go with a separate template for each board, but I do agree that we might need multiple templates for very diverging designs.  But I would say that, for example, every board supported by the new `arduino-hal` from #117 can be supported from a single template.

I saw that `cargo-generate` is making progress on their issue https://github.com/cargo-generate/cargo-generate/issues/17, so I think we can make use of that in the near future :)

Rahix, 2021-03-13:
For reference: The template should contain a `rust-toolchain` file (https://github.com/Rahix/avr-hal/issues/124#issuecomment-795585670 and [the toolchain file docs](https://rust-lang.github.io/rustup/overrides.html#the-toolchain-file)).

Rahix, 2021-11-01:
<https://github.com/Rahix/avr-hal-template>

closed 2021-11-01 by Rahix