Rahix/avr-hal #70: TTY not showing up on arduino Micro

TerminatorNL, 2020-09-13:
## Foreword
First off, I am a huge fan! Thank you for putting work into this project!
I am relatively new to Arduino programming, so this entire post might be a fault of of my own.

## General info
- Device: `Arduino Micro`
- OS: `Fedora 32`, running linux kernel 5.5.6
- I followed the `Arduino Leonardo` example, since the Arduino Micro shares the same microchip (`ATmega32U4`)

## The issue at hand: Serial monitor unavailable
When I use Arduino IDE to flash a bare bones program to the arduino (Appendix 1 & 2) I get a tty device under `/dev/ttyACM0` regardless of the program contents.

When I use avrdude to flash the rust program (Appendix 3 & 4) no tty device is created. I also checked `/dev/ttyS0`, it exists, but does not generate contents.

This means I am unable to view the output of the device since the serial monitor has nothing to connect to. I'm not sure what other diagnostics to perform, but I'm willing to supply more information if needed!

Thanks a bunch for creating this project, I can't wait to use it!

## Appendices
### Appendix 1: Code within arduino IDE
```c
void setup() {
  
}

void loop() {
  delay(1);
}
```

### Appendix 2: Arduino IDE and avrdude output
<details>
  <summary>Spoiler: Arduino IDE with blank script: avrdude</summary>

```
/home/predator/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/predator/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega32u4 -cavr109 -P/dev/ttyACM0 -b57600 -D -Uflash:w:/tmp/arduino_build_696973/DigitalReadSerial.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/predator/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/predator/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
[... 132 lines cut ...]
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file "target/avr-atmega32u4/release/avr-example.elf"
avrdude: writing flash (516 bytes):

Writing | ################################################## | 100% 0.06s

avrdude: 516 bytes of flash written
avrdude: verifying flash memory against target/avr-atmega32u4/release/avr-example.elf:
avrdude: load data flash data from input file target/avr-atmega32u4/release/avr-example.elf:
avrdude: input file target/avr-atmega32u4/release/avr-example.elf contains 516 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 516 bytes of flash verified

avrdude done.  Thank you.
```
  
</details>

TerminatorNL, 2020-09-13:
Extra clarification: If I flash a program written with Arduino IDE I can connect to the serial monitor. For some reason, I can't do it using avr-hal generated scripts. I put Arduino IDE in the description above to show that regardless of program contents, the TTY is active, but that is not the case with my avr-hal script.

TerminatorNL, 2020-09-13:
Extra clarification 2: The script that's generated with avr-hal DOES WORK. The LED turns on after upload.

Rahix, 2020-09-14:
Hi @TerminatorNL,

you're trying to access the serial console via USB, right?  That doesn't work in Rust yet as the USB stack has not yet been implemented.  What's going on is this:  Arduino sketches under the hood initialize the USB stack and expose a CDC-ACM device.  That's what you see on your computer and what the  serial monitor connects to.

The `arduino_leonardo::Serial` in Rust, however, is the actual hardware serial interface, connected to pins D0, D1.  On the Arduino Uno, that's wired via a USB-to-Serial converter to the USB port.  This way, writing to the hardware serial can be read on the computer.  On Arduino Leonardo/Micro, the USB port is directly wired to the MCU (ATmega32U4) and the hardware serial is only connected to pins D0 and D1.

So, to use serial, you currently have to connect an external USB-to-Serial converter to pins D0, D1 on Arduino Leonardo/Micro.  Once USB support is implemented and ready (issue #40), we can add a seconds serial interface to the `arduino-leonardo` crate that does the same thing as Arduino sketches (initialize a CDC-ACM device which a computer connected via USB can read from/write to).

Hope this makes sense?

TerminatorNL, 2020-09-14:
This makes total sense! Thank you so much for your kind and in-depth explanation! Sadly I don't have a USB to serial converter right now. I'll probably buy one in the near future.

My current project requires USB usage (that will definitely require #40 ), which means I will have to wait a bit.

I will definitely re-visit this project once USB gets implemented!

closed 2020-09-14 by TerminatorNL