Rahix/avr-hal #71: Tests fail

ToddG, 2020-09-18:
HASH
commit ad6fedd35a9a17639edf44a59fab6a14b2707cce (HEAD -> master, origin/master, origin/HEAD)
Author: Rahix <rahix@rahix.de>
Date:   Mon Sep 7 23:06:02 2020 +0200

    generic: delay: Fix a compiler warning
    
    Remove unnecessary parentheses.
    
    Signed-off-by: Rahix <rahix@rahix.de>

REPRO
git clone https://github.com/Rahix/avr-hal
cd avr-hal
cargo build
cargo test

ERROR
attached.
[cargo-test-out.txt](https://github.com/Rahix/avr-hal/files/5246624/cargo-test-out.txt)

QUESTION
Why aren't the tests being run as part of the travis build? I only see `cargo build` being run.

ToddG, 2020-09-18:
Posted a related bug/question on SO here:

https://stackoverflow.com/questions/63961435/rust-cargo-test-fails-for-arduino-targets-with-duplicate-lang-item-in-crate

Rahix, 2020-09-18:
There's a few things here ...

First of all, what do you expect `cargo test` to do?  The crates in this repo target a microcontroller, it's hard to test those without actual hardware ;) So it doesn't really make sense to run `cargo test`, there isn't much it could test and no way to run AVR machine code on your computer without an emulator anyway ...

If you take a look at the code base, there aren't any `#[test]`s defined anywhere to run.  The tests are (more or less) the examples for each board as that's the best thing we have to test functionality at this point.

In general, `cargo test` doesn't support testing on embedded targets yet.  There are frameworks like [`utest`](https://github.com/japaric/utest) which try to make it work but that's all really experimental at this point.

That said, regarding the errors you've seen:

```text
error: Ensure that you are using an AVR target! You may need to change directories or pass a --target flag to cargo. See [...]
```
Well, the message says it all: You didn't specify an AVR target with `--target` so rust tried to build `avr-device` for your computer which doesn't make sense - hence the error.  When building inside a board directory, the `.cargo/config.toml` file specifies the target for you; that's why you didn't get the above error there ...

```text
error: duplicate lang item in crate `core` (which `rustc_std_workspace_core` depends on): `bool`.
```
I don't know for certain, but I would assume this error is a symptom of the facts that A) `cargo test` does not support bare-metal targets and B) std aware cargo (i.e. building custom libcore) is very experimental and not yet fully implemented.

ToddG, 2020-09-18:
Thank you for the detailed message. This is my first brush with rust, so here we go. I'm using `avr-hal` as a base case for understanding how the `cargo` commands _should_ work.

I was expecting that since workspace commands at the top level such as `cargo clean` and `cargo build` work by iterating over the member crates, then `cargo test` would, too. With no #[test] blocks, I'd expect the workspace command to iterate through the member crates and report back zero tests. At any rate, I would not expect to get a bunch of build errors.

I think the AVR target message is a red herring. The top level `cargo` commands invoke `cargo` commands in the child crates, and those crates have the necessary `.cargo/config.toml` files. 

Q: How do people test embedded targets like the AVR in rust? After spending a full day trying to get `cargo test` running (nowhere in the documentation have I seen mention that this is unsupported)...I've decided to just write my code and tests in a standard workspace, and then copy and paste the code (without the tests) into the .rs files within a crate that targets the AVR.

Thanks again.

Rahix, 2020-09-19:
> I'm using avr-hal as a base case for understanding how the cargo commands should work.

Hm, you might want to look at something else first, `avr-hal` and similar projects are quite special and probably not the best way to understand cargo ... Especially `avr-hal` because it uses unstable cargo features.

> I was expecting that since workspace commands at the top level such as cargo clean and cargo build work by iterating over the member crates, then cargo test would, too. With no #[test] blocks, I'd [...]

The build errors were because it is fundamentally impossible to run AVR machine code on your computer and because `cargo test` does not support bare-metal targets.  I'd say for `avr-hal` there is not other sensible outcome from running `cargo test` than a build error; at least at the moment ...

> I think the AVR target message is a red herring. The top level cargo commands invoke cargo commands in the child crates, and those crates have the necessary .cargo/config.toml files.

No, `.cargo/config.toml` is only pulled in from the directory you're running the command from (and any directory above).  So `cargo ...` in the top level directory will _not_ have any target or other options set.  That's the reason the cargo config is separate from `Cargo.toml`, because it contains the parts that should not be crate specific.

> Q: How do people test embedded targets like the AVR in rust? After spending a full day trying to get cargo test running (nowhere in the documentation have I seen mention that this is [...]

It really depends on what you're trying to test:  For unit-testing isolated, machine-independent, parts people usually put those in a separate library crate that they can then run tests on on their development machine.  For anything machine specific, things you can only really test on the target device, it's really project-specific which methods make sense:  You could add a feature-flag to include selftest routines in the firmware build, have a second binary crate that calls into the machine-specific functionality to test it, try using crates like `utest` as I mentioned before, etc.  There isn't really a canonical way and, to be quite honest, automated testing is not (yet) that far spread in the embedded (not just Rust) community in general ...

Rahix, 2020-10-02:
Is this issue resolved for you now?

ToddG, 2020-10-02:
Yes, this is resolved. Thank you.

-------- Original Message --------

closed 2020-10-02 by Rahix