Rahix/avr-hal #75: Need examples showing how to use timer interrupts : ISR : OCR: interrupt service routine: output compare register

ToddG, 2020-09-25:
## BACKGROUND

I'm attempting to follow various examples for writing timer driven interrupt service routines in rust targeting the AVR...similar to these examples in `arduino c`:

https://www.instructables.com/id/Arduino-Timer-Interrupts/
https://circuitdigest.com/microcontroller-projects/arduino-timer-tutorial

### EXAMPLE `c` CODE

In these examples, there is a stanza for setting the various registers to set this up. Here's an extract:

```c
//set timer1 interrupt at 1Hz
  TCCR1A = 0;// set entire TCCR1A register to 0
  TCCR1B = 0;// same for TCCR1B
  TCNT1  = 0;//initialize counter value to 0
  // set compare match register for 1hz increments
  OCR1A = 15624;// = (16*10^6) / (1*1024) - 1 (must be <65536)
  // turn on CTC mode
  TCCR1B |= (1 << WGM12);
  // Set CS10 and CS12 bits for 1024 prescaler
  TCCR1B |= (1 << CS12) | (1 << CS10);  
  // enable timer compare interrupt
  TIMSK1 |= (1 << OCIE1A);
```

Then these c examples show how to wire up an ISR:

```c
ISR(TIMER1_COMPA_vect){...}
```

### AVR-HAL `rust` CODE

In the `avr-hal` repo's various examples, I only see reference to the timer registers of `tccr1b` and `tcnt1`.  Is this an oversight or is this because of some limitation on the platform? It would be really useful to see an example that included the other registers such as compare `ocr1a` and especially an example showing the use of an interrupt service routine like the common `c` ISR above.

Links:
Here's the example code I am following, though I've looked at all the `avr-hal` examples that include a timer:

https://github.com/Rahix/avr-hal/blob/2a59d441a5b9506b3cbfac21e532639733a6314a/boards/arduino-uno/examples/uno-hc-sr04.rs#L22

From that example, we can see where wi set TCCR1B:

_tccr1b_
```rust
 let timer1 = dp.TC1;
    // Starting and initializing the timer with prescaling 64.
    // it gives one clock count every 4 µs.
    // since the clock register size is 16 bits, the timer is full every
    // 1/(16e6/64)*2^16 ≈ 260 ms
    timer1.tccr1b.write(|w| w.cs1().prescale_64());
```

Here the TCNT1 is reset:

_tcnt1_
```rust
        timer1.tcnt1.write(|w| unsafe { w.bits(0) });
```

And is this the equivalent of an output compare register (OCR) read?

_tcnt1_
```rust
            if timer1.tcnt1.read().bits() >= 50000 {
```

## REQUEST

An example that actually sets up 2 different timers, say timer0 and timer1, and instantiates an ISR. I assume this would mean properly setting the `ocr0a` and `ocr1a` registers...

ToddG, 2020-09-26:
I have gotten multiple timers working by manually doing the compare within the loop like so:

https://github.com/ToddG/learn-electronics/blob/9d4c7b32a10a9d2063966cbe76da276b1274d1b4/posts/hello-world/07-multisegment-led-rust-arduino/code/multisegment-led/avr/mseg-bin/src/main.rs#L183
```rust
// enter loop
    loop {
        // manual compare timer1 : 1Hz
        if timer1.tcnt1.read().bits() >= 62499 {
            // --------------------------------------------------
            // DO SOMETHING AS IF THIS WERE AN ISR FOR TIMER1
            // --------------------------------------------------
            hardware = handle_timer1_1hz(hardware);
            // reset timer
            timer1.tcnt1.write(|w| unsafe { w.bits(0) });
        }
        // manual compare timer0 : 1kHz
        if timer0.tcnt0.read().bits() >= 249 {
            // --------------------------------------------------
            // DO SOMETHING AS IF THIS WERE AN ISR FOR TIMER0
            // --------------------------------------------------
            hardware = handle_timer0_1k_hz(hardware);
            // reset timer
            timer0.tcnt0.write(|w| unsafe { w.bits(0) });
        }
    }
}
```

This strategy works fine, though I still wonder if there isn't a more elegant way to configure the ISRs.

kallemooo, 2020-09-27:
This is a quite new RUST HAL-layer for AVR and all features is not yet implemented.
Help is always wanted.

For delays using timers the issue #9 is planned.

Rahix, 2020-10-02:
Before divining into this, a fair warning: Dealing with interrupts is _hard_.  Even in rust, where shooting yourself in the foot is prevented in most cases, you have to be careful.  And you will need unsafe in places so you need to understand very well what you are doing.

Okay, that said, first of all the entire code (this one is for Arduino Leonardo, but I'm sure you can adapt it to whatever board you're using):

<details>
<summary>Click to expand entire listing</summary>

```rust
#![no_std]
#![no_main]
#![feature(abi_avr_interrupt)]

use arduino_leonardo::hal::port;
use arduino_leonardo::prelude::*;
use core::mem;
use panic_halt as _;

static mut LED: mem::MaybeUninit<port::portc::PC7<port::mode::Output>> = mem::MaybeUninit::uninit();

#[arduino_leonardo::entry]
fn main() -> ! {
    let dp = arduino_leonardo::Peripherals::take().unwrap();

    let mut pins = arduino_leonardo::Pins::new(dp.PORTB, dp.PORTC, dp.PORTD, dp.PORTE, dp.PORTF);

    let led = pins.d13.into_output(&mut pins.ddr);
    unsafe {
        // SAFETY: Interrupts are not enabled at this point so we can safely write the global
        // variable here.  A memory barrier afterwards ensures the compiler won't reorder this
        // after any operation that enables interrupts.
        LED = mem::MaybeUninit::new(led);
        core::sync::atomic::compiler_fence(core::sync::atomic::Ordering::SeqCst);
    }

    // Timer Configuration:
    // - WGM = 4: CTC mode (Clear Timer on Compare Match)
    // - Prescaler 256
    // - OCR1A = 15624
    //
    // => F = 16 MHz / (256 * (1 + 15624)) = 4 Hz
    //     (^ this formula I deduced from reading the datasheet)
    //
    // => The LED will toggle at 4 Hz, thus it blinks at 2 Hz
    let tmr1 = dp.TC1;
    tmr1.tccr1a.write(|w| w.wgm1().bits(0b00));
    tmr1.tccr1b.write(|w| w.cs1().prescale_256().wgm1().bits(0b01));
    tmr1.ocr1a.write(|w| unsafe { w.bits(15624) });

    // Enable the timer interrupt
    tmr1.timsk1.write(|w| w.ocie1a().set_bit());

    // In theory this should not be necessary ... But if you previously had
    // a sketch from Arduino loaded, the USB device will not have been reset.
    // Because of this we will be spammed with interrupts which will never
    // stop because they are never handled.
    //
    // (only for ATmega32U4)
    dp.USB_DEVICE.usbcon.reset();

    // Enable interrupts globally
    unsafe {
        // SAFETY: Not inside a critical section and any non-atomic operations have been completed
        // at this point.
        avr_device::interrupt::enable();
    }

    loop {
        avr_device::asm::sleep();
    }
}
[... 63 lines cut ...]
    avr_device::asm::sleep();
}
```

Because the main program has nothing to do in this case, I just let it sleep until any interrupt happens (and then make it immediately go back to sleep again).

```rust
#[avr_device::interrupt(atmega32u4)]
fn TIMER1_COMPA() {
    // ...
}
```

This is how to actually define an ISR, similar to the `ISR()` C macro.  The function name _must_ be the name of a know interrupt.  You can look them up here: [for ATmega32U4 (Arduino Leonardo)][isrs-32u4] or [for ATmega328P (Arduino Uno)][isrs-328p].

[isrs-32u4]: https://docs.rs/avr-device/0.2.2/avr_device/atmega32u4/enum.Interrupt.html
[isrs-328p]: https://docs.rs/avr-device/0.2.2/avr_device/atmega328p/enum.Interrupt.html

The `avr_device::interrupt` macro also needs the name of the CPU and its important to use the right one here.  Otherwise you'll get strange results ...

There are a few important things to remember when writing ISRs:

- As long as the ISR is running, _nothing_ else can happen.  The main program is stopped and _no_ other interrupt can trigger.  Thus, **do as little as possible**; try to offload any actual work into the main program and just set a signal flag in the ISR.
- The ISR can run at _any_ time when interrupts are enabled.  You absolutely _must_ synchronize access to shared data (using critical sections).
- An interrupt firing can usually mean that multiple hardware events might have occured.  You should be prepared to handle them all instead of assuming you'll get one interrupt for one event.  For example, if interrupts were disabled for a long time, a timer might have overflown multiple times.

ToddG, 2020-10-05:
Thank you for the detailed write up. I'm beginning to see what you mean about this being complicated. If nothing else, this is useful to force me to learn various aspects of Rust and embedded development.

I have extended my example to follow your advice w/respect to guarding state, etc. See [full example](https://github.com/ToddG/learn-electronics/tree/master/posts/hello-world/08-rust-timers) for context, but here's the snippet that is currently blocking me. Basically, I want to encapsulate the board/device pins in a global struct that guards access with a critical section. However, I am having a tough time understanding the macro expansion of the `Pins`. Here are my [specific comments from this code](https://github.com/ToddG/learn-electronics/blob/29d06244d836a7bf10dde507e8a2541ae661be72/posts/hello-world/08-rust-timers/multisegment-led/mseg-bin/src/main.rs#L30)

```rust
/*
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
This does not work. The problem is...how do I create a default Output Pin
to populate this global/static variable? I can set it in the entrypoint,
but I need to allocate the space here, as I do for the other globals...
Specifically:
    Pin::<mode::Output>::new()
I don't see how to create a new generic output pin.
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
struct WrappedHardware(UnsafeCell<Hardware>);
unsafe impl Sync for WrappedHardware {}
static WRAPPED_HARDWARE: WrappedHardware =
    WrappedHardware(
        UnsafeCell::new(
            Hardware{
                    onboard_led_pin: Pin::<mode::Output>::new(),
                    pins: arr![Pin::<mode::Output>::new(); 10] }));
 */
```

Rahix, 2020-10-06:
> I'm following a pattern of encapsulating shared state in an UnsafeCell and then guarding access with critical sections. I'm doing this for: such as LedToStrobe, DisplayCounter, and MultiplexedLEDArray.

First of all, you're missing compiler fences here.  You should add a

```rust
core::sync::atomic::compiler_fence(core::sync::atomic::Ordering::SeqCst);
```

to the end of each critical section to ensure that operations won't get reordered.

> I don't see how to create a new generic output pin.

You can't.  The pins must be taken from the `Pins` struct and moved into your global.  If there was some kind of `Pin::new()` function, this would be unsound, as it would allow creating multiple variables for the same pin.

To create generic pins from the concrete pin types you find in the [`Pins`](https://rahix.github.io/avr-hal/arduino_uno/struct.Pins.html) struct, you should use the `.downgrade()` method.  This will give you a variable of the generic `Pin` type.  Read more about this in the [Digital I/O Documentation](https://rahix.github.io/avr-hal/avr_hal_generic/port/index.html#downgrading).  As an example, [`leonardo-blink.rs`](https://github.com/Rahix/avr-hal/blob/master/boards/arduino-leonardo/examples/leonardo-blink.rs) uses this to store three pins in an array.

ToddG, 2020-10-09:
@Rahix Thank you for the additional comments.

I spent 2 full days trying to get timers to work. For anyone following, let me just state that this will drive you insane. The timers are incredibly fragile. I have lost count of the ways that I have broken them... and debugging by just looking at a few blinking leds is pretty draining as well. 

For anyone interested, I'll paste in my code but I'm not going to spend anymore time on this. Keeping the code in the main thread and just checking the timer values worked well enough.

main.rs
```rust
#![no_std]
#![no_main]
#![feature(abi_avr_interrupt)]

use arduino_uno::hal::port;
use arduino_uno::prelude::*;
use arr_macro::arr;
use core::cell::UnsafeCell;
use core::mem;
use panic_halt as _;
use avr_device::interrupt;
use mseg_lib::{cmap, bits};

/// ------------------------------------------------------------------------
/// Consts
/// Modify these values based on the number of LEDs:
/// ------------------------------------------------------------------------
/// MAX_VALUE : if 1 LED, then this would be 10, 2 leds then 100, 3 leds then 1000.
/// This is obviously pow(10, NUM_LEDS).
const MAX_VALUE: usize = 100;

/// The number of LEDS in the display. Common numbers are 1, 2, 6, etc. Whatever
/// you have wired up.
const NUM_LEDS: usize = 2;

/// Always gonna be 10
const BASE: usize = 10usize;

/// This is where you wire up the LED segments to the GPIO pins.
const LEDS_COMMON_CATHODES: [usize; 8] = [2, 3, 4, 5, 6, 7, 8, 9];
const LED_ANODES: [usize; 2] = [10, 11];

/// ------------------------------------------------------------------------
/// State
/// ------------------------------------------------------------------------
#[derive(Debug, Clone, Copy)]
struct State {
    _led_to_strobe_index: usize,
    _display_counter: usize,
}

impl State {
    pub fn increment_led_to_strobe(&self) -> usize {
        (self._led_to_strobe_index + 1) % NUM_LEDS
    }
    pub fn increment_counter(&self) -> usize {
        (self._display_counter + 1) % MAX_VALUE
    }
    pub fn led_to_strobe(&self) -> usize {
        self._led_to_strobe_index
    }
    pub fn counter(&self) -> usize {
        self._display_counter
    }
    pub fn digits(&self) -> [u8; NUM_LEDS] {
        let mut num = self._display_counter;
        let mut result = [0; NUM_LEDS];
        // extract each digit from the count up counter and update the display value accordingly
        for i in 0..NUM_LEDS {
            result[i] = (num % BASE) as u8;
            num /= BASE
        }
[... 334 lines cut ...]
    ///  Hardware Pins
    ///      A  B   C   D   E   F   G   DP  COM
    ///      7  6   4   2   1   9   10  5   3/8
    ///
    /// In the array, the pins are indexed as:
    ///      A  B   C   D   E   F   G   DP  COM
    ///      0  1   2   3   4   5   6   7   8
#[derive(Clone, Copy, Debug)]
pub struct EightSegmentLEDCommonAnode {
    // pins (0-7 are A-DP, 8 is com)
    pub pins: [usize; 9],
}

impl EightSegmentLEDCommonAnode {
    pub fn pins(&self) -> [usize;9] {
        self.pins
    }
    pub fn com(&self) -> usize {
        self.pins[8]
    }
}
```
Links
* https://docs.rs/rust-libcore/0.0.3/core/sync/atomic/fn.compiler_fence.html
* https://doc.rust-lang.org/core/sync/atomic/fn.compiler_fence.html

Rahix, 2020-10-09:
> I spent 2 full days trying to get timers to work. For anyone following, let me just state that this will drive you insane. The timers are incredibly fragile. I have lost count of the ways that I have broken them... and debugging by just looking at a few blinking leds is pretty draining as well.

Oh, I know this all too well!  Props to you for going this far though!

> For anyone interested, I'll paste in my code but I'm not going to spend anymore time on this. Keeping the code in the main thread and just checking the timer values worked well enough.

Just a few comments about your code and regarding this:  Doing all the work in ISRs is really not ideal.  AVR does not have interrupt nesting so during an ISR absolutely nothing can happen in the system.  Ideally, ISRs should be as short as possible and any work should happen in the main thread.  This also helps avoid all the nasty synchronization issues ... A common pattern is a global flag marker that is set by the ISR and reset by the main thread:

```rust
(not tested)
use core::sync::atomic;

static TMR_OVERFLOW: atomic::AtomicBool = atomic::AtomicBool::new(false);

#[avr_device::interrupt(atmega328p)]
fn TIMER1_COMPA() {
    TMR_OVERFLOW.store(true, atomic::Ordering::SeqCst);
}

#[arduino_uno::entry]
fn main() -> ! {
    // ...

    loop {
        let tmr_overflow = avr_device::interrupt::free(|_cs| {
            // With interrupts disabled, check and reset the flag.  If interrupts
            // were enabled, the ISR could run in between the check and reset, leading to
            // us loosing a timer overflow.
            if TMR_OVERFLOW.load(atomic::Ordering::SeqCst) {
                TMR_OVERFLOW.store(false, atomic::Ordering::SeqCst);
                true
            } else {
                false
            }
        });

        if tmr_overflow {
            // the code you currently have in the ISR goes here.
        }
    }
}
```

ToddG, 2020-10-21:
Related question, in that using I2C is going to use interrupts under the covers (I think).

Q1: How to write the code for a `slave` that listens on an address?

Both the sample code and the I2C API docs seem to be from the perspective of the master:
* https://rahix.github.io/avr-hal/atmega328p_hal/prelude/trait._embedded_hal_blocking_i2c_Read.html

For context, I am attempting to implement a simple master/slave example from here: http://gammon.com.au/i2c

Rahix, 2020-10-21:
> Related question, in that using I2C is going to use interrupts under the covers (I think).

No, the I2C (also called TWI an AVR MCUs) master driver currently does not enable any interrupts at all.  Slave operation should also be possible without interrupts.  Though in practice, usually you'd want the core to go to sleep while waiting for a bus transaction, which of course will need to make use of an interrupt.

> Both the sample code and the I2C API docs seem to be from the perspective of the master:

There is no I2C slave driver in `avr-hal` as of now.  You'd need to manually do this with direct register accesses (or write your own I2C slave driver).

Rahix, 2020-10-21:
> There is no I2C slave driver in `avr-hal` as of now. You'd need to manually do this with direct register accesses (or write your own I2C slave driver).

If you are really interested in this we could work together on adding an I2C slave driver to `avr-hal` though.  Just let me know!

ToddG, 2020-10-21:
Absolutely, let's do this! Ok, so I'll start brainstorming:

* Are there other platforms that we could look at and model after?
* What sections of the AVR spec should I look at closely?
* What's the overall game plan?

My sense of this is that the slave needs to:

* set an interrupt on the SDA pin being pulled low
* the interrupt wakes the CPU and sets a flag
* main thread loop checks flag, if set, invokes reader function with user defined callback
* callback is given an iterator over a stream

Conceptually, it looks simple enough when looking at Arduino code: ```Wire.onReceive (receiveEvent);```

Rahix, 2020-10-21:
Right, let's move this to a separate issue though: #90

closed 2020-10-26 by Rahix