Rahix/avr-hal #77: Switch to embedded-hal 1.0.0

Rahix, 2020-10-02:
Not sure when is a good time to do this but it would be great to be an early adopter of the new traits :) They bring a lot of improvements to ergnomics and usability which would help the readability of downstream code a lot.

We do, however, have to consider that most drivers won't have updated yet so we have to be careful as to not become 100% incompatible.  Maybe there is a possibility for a compat-shim layer or a cargo-feature to allow using the older version in parallel?

Rahix, 2020-10-26:
For reference, some other HALs have gone the route of supporting both the release and alpha versions of `embedded-hal` at the same time:

https://github.com/lpc-rs/lpc8xx-hal/blob/6e787f8e26d908b3cea13c3c47f04695f4112837/Cargo.toml#L40-L46

If done carefully, I could see this bringing nice benefits.  We could have the HAL API look nice because of the alpha traits but still keep compatibility with 'old' drivers through the release version.

stappersg, 2025-01-01:
For what it is worth, merge request #621 is titled _"[arduino-hal] remove embedded-hal-v0 dependency"_.

Rahix, 2025-01-04:
This issue was mostly taken care of in #488 which unfortunately never got linked back here.  There were some additional PRs like #517, but I think by now the transition can be considered completed.

closed 2025-01-04 by Rahix