Rahix/avr-hal #80: More fine grained control of serial baudrate

Rahix, 2020-10-04:
Currently, the code only allows specifying a baudrate which the HAL-driver will try to match as closely as possible.  The "generic" algorithm used for this will not always produce the most optimal settings, however.  This means, for example, that 115200 baud on Arduino Uno is completely broken.

It would be nice to also allow setting exact baudrate parameters or/and providing alternate algorithms.  There are quite a few ways to implement this, here are a few ideas:

1. A new `Baudrate` type with methods for the different modes:

   ```rust
   // Clock is needed because the calculations needs to take core clock into account
   pub struct Baudrate<CLOCK> {
       pub ubrr: u16,
       pub u2x: bool,
       pub _clock: PhantomData<CLOCK>,
   }

   impl<CLOCK: crate::clock::Clock> Baudrate<CLOCK> {
       pub fn new(baud: u32) -> Baudrate<CLOCK> {
           // current implementation
       }

       pub fn with_exact(u2x: bool, ubrr: u16) -> Baudrate<CLOCK> {
           Baudrate { ubrr, u2x, _clock: PhantomData }
       }
   }
   ```

   The `Usart` constructor now takes `Baudrate<CLOCK>` instead of `u32`.

2. Using an `Into<Baudrate<CLOCK>>` bound in the `Usart` constructor.  We also implement `From<u32> for Baudrate<CLOCK>`.  This would allow backwards compatibility with current code (but I'm not sure if that is a good thing ...).

explicite, 2020-10-10:
I will try to start this. Probably I will have lot of questions...
1. This is the `generic algorithm` mentioned by you `let brr = CLOCK::FREQ / (16 * baud) - 1;`?
1. `Baundrate` should be defined as distinct file under generic src like Clock?

Rahix, 2020-10-11:
> This is the generic algorithm mentioned by you let brr = CLOCK::FREQ / (16 * baud) - 1;?

Yes, it's this code:

https://github.com/Rahix/avr-hal/blob/bad92776522bbbedee45d276991b45fb81f253d8/avr-hal-generic/src/serial.rs#L56-L58

In comparison, Arduino upstream uses [this code](https://github.com/arduino/ArduinoCore-avr/blob/3055c1efa3c6980c864f661e6c8cc5d5ac773af4/cores/arduino/HardwareSerial.cpp#L119-L136):

```cpp
  // Try u2x mode first
  uint16_t baud_setting = (F_CPU / 4 / baud - 1) / 2;
  *_ucsra = 1 << U2X0;

  // hardcoded exception for 57600 for compatibility with the bootloader
  // shipped with the Duemilanove and previous boards and the firmware
  // on the 8U2 on the Uno and Mega 2560. Also, The baud_setting cannot
  // be > 4095, so switch back to non-u2x mode if the baud rate is too
  // low.
  if (((F_CPU == 16000000UL) && (baud == 57600)) || (baud_setting >4095))
  {
    *_ucsra = 0;
    baud_setting = (F_CPU / 8 / baud - 1) / 2;
  }

  // assign the baud_setting, a.k.a. ubrr (USART Baud Rate Register)
  *_ubrrh = baud_setting >> 8;
  *_ubrrl = baud_setting;
```

> Baundrate should be defined as distinct file under generic src like Clock?

No, I'd prefer putting that in `avr-hal-generic/src/serial.rs` as it is very closely tied to it.

> Probably I will have lot of questions...

Sure, please don't hesitate asking if anything is unclear!

closed 2020-11-07 by Rahix