Rahix/avr-hal #86: Runner support for macOS

explicite, 2020-10-09:
Leonardo runner in macOS require some additional changes. 
1. `numfmt` installation `brew install coreutils`
2. `avrdude` installation `brew install avrdude`
3. `avrdude` conf location update
4. `avrdude: ser_open(): can't open device "/dev/ttyACM0": No such file or directory` after first upload

@Rahix do you have any idea how we can improve this?

Rahix, 2020-10-09:
I want to replace the runners with a proper tool at some point but that's going to be a bigger change.

For now, I'm not sure.  I think the easiest way is to just hack the script locally ... If it's not too difficult, we could also add some more intelligent behavior to it, for example:

- skipping human size display if numfmt isn't installed
- choosing the macos serial device name & avrdude config location if running on a macos system

Rahix, 2020-10-13:
Implemented in #87 and commit 7337cd76cd96f2d27701b137396d94a06d3a501d.

closed 2020-10-13 by Rahix