Rahix/avr-hal #91: Firmware size too large

Nexyll, 2020-10-21:
Hi,
I initialized a new project for the Arduino Nano board following the startup guide but for the blink example I get a firmware of more than 32kb once compiled.
In the README I understood that the Arduino Nano / UNO were supported but as their flash memory is 32kb I have to do something backwards.

I use this command (under windows) to upload the firmware on the board : `avrdude -patmega328p -carduino -PCOM3 -b57600 -D -Uflash:w:avr-example.elf`

Here are my source files :

**main.rs**

```rust
#![no_std]
#![no_main]

extern crate panic_halt;

use arduino_uno::prelude::*;

#[arduino_uno::entry]
fn main() -> ! {
    let dp = arduino_uno::Peripherals::take().unwrap();
    let mut pins = arduino_uno::Pins::new(dp.PORTB, dp.PORTC, dp.PORTD);
    let mut led = pins.d13.into_output(&mut pins.ddr);

    loop {
        led.set_high().void_unwrap();
        arduino_uno::delay_ms(1000);
        led.set_low().void_unwrap();
        arduino_uno::delay_ms(1000);
    }
}
```

**cargo.toml**

```toml
[package]
name = "avr-example"
version = "0.1.0"
authors = ["Nexyll <antoine7104@gmail.com>"]
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
# A panic handler is needed.  This is a crate with the most basic one.
# The `leonardo-panic` example shows a more elaborate version.
panic-halt = "0.2.0"

[dependencies.arduino-uno]
git = "https://github.com/Rahix/avr-hal"
rev = "0c6cf1675c2724354f1adeaeee69992acd371e80"
feature = ["arduino-nano"]

# Configure the build for minimal size
[profile.dev]
panic = "abort"
lto = true
opt-level = "s"

[profile.release]
panic = "abort"
codegen-units = 1
debug = true
lto = true
opt-level = "s"
```

**config.toml**

```toml
[build]
target = "./avr-atmega328p.json"

[target.'cfg(target_arch = "avr")']
runner = "./uno-runner.sh"

[unstable]
build-std = ["core"]
```

Rahix, 2020-10-21:
Hmm, that seems very wrong ... I see a size of ~500 bytes for the blink examples:

```text
❯ avr-size ../../target/avr-atmega328p/debug/examples/uno-blink.elf 
   text	   data	    bss	    dec	    hex	filename
    530	      0	      1	    531	    213	../../target/avr-atmega328p/debug/examples/uno-blink.elf
```

Can you run `avr-size` on your binary to check what size it really is?  What size do you see when building the examples in `avr-hal`?  What compiler version are you using?

Nexyll, 2020-10-21:
I must have missed something.

When i run the `avr-size` command against the .elf file i got

```
> avr-size avr-example.elf
   text    data     bss     dec      hex     filename
   360       0       1         361     169     avr-example.elf
```
So it seems to be pretty the same as on your side. I use this version of avr-gcc : `avr-gcc (WinAVR 20100110) 4.3.3`

Maybe It's during the upload with avr-dude that the problem occur?

```
> avrdude -patmega328p -carduino -PCOM3 -b57600 -D -Uflash:w:avr-example.elf

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "avr-example.elf"
avrdude: input file avr-example.elf auto detected as raw binary
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 9.93s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against avr-example.elf:
avrdude: load data flash data from input file avr-example.elf:
avrdude: input file avr-example.elf auto detected as raw binary
avrdude: input file avr-example.elf contains 32768 bytes <----------------------- HERE
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 7.72s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x7800
         0x2e != 0x0c
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK

avrdude done.  Thank you.
```

Rahix, 2020-10-21:
> `avrdude: input file avr-example.elf auto detected as raw binary`

this does not seem right ... We're using the following avrdude invocation in `uno-runner.sh`:

https://github.com/Rahix/avr-hal/blob/0c6cf1675c2724354f1adeaeee69992acd371e80/boards/arduino-uno/uno-runner.sh#L58

I think you might be missing the `:e` at the end?

Nexyll, 2020-10-22:
I had deleted this param because I was getting the following error under windows : 
```
> avrdude -patmega328p -carduino -PCOM3 -b57600 -D -Uflash:w:avr-example.elf:e
avrdude: invalid file format 'e' in update specifier
avrdude: error parsing update operation 'flash:w:avr-example.elf:e'
```

Finally, with this command I generated a .hex which then passes into avr-dude and it works
```
> avr-objcopy -O ihex avr-example.elf avr-example.hex
```

Thanks again for your help !

Rahix, 2020-10-22:
Interesting ... Maybe you have an old version of `avrdude` (I have 6.3)?  Or maybe ELF support is not compiled into your version for some reason?

closed 2020-10-26 by Rahix

Nexyll, 2020-11-09:
Yeah, i use this version : `avrdude version 5.10` maybe quite outdated...

bacongobbler, 2020-11-14:
Just wanted to chime in with my own experience. It's because WinAVR is still on version 5.10. I'm using WinAVR because WSL2 does not have serial support at this time. Using @Nexyll's workaround in https://github.com/Rahix/avr-hal/issues/91#issuecomment-714649530 worked great. Thank you for finding that!

It is worth noting that the last update to WinAVR was in 2010... Definitely not ideal. If anyone has a more updated and up-to-date solution for those looking for `avrdude` on Windows... I'm all ears.

https://sourceforge.net/projects/winavr/files/

lights0123, 2020-11-14:
@bacongobbler the Arduino IDE ships with 6.3-20190619

bacongobbler, 2020-11-14:
Oh wonderful! Thank you for the pointer @lights0123.

For those following along, that can be found via

```
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avrdude.exe
```