Rahix/avr-hal #99: Some uncommon baudrates might be unstable

Rahix, 2020-11-06:
Following PR #88, some (uncommon!) baudrates seem to be unstable:  Namely 14400, 28800, and 76800 seem to be problematic.  Further analysis of the generated waveform is needed to determine what is the problem.

Cc: @explicite

explicite, 2020-11-07:
@Rahix I want to create test for this. I'm not sure if it's possible. Let me explain my idea.

I want to connect **mega2560** pins 18 and 19 to **Uno** on with problematic baudrates**. Uno** will only echo char from from **mega2560**.
**mega2560** will be connected via usb to ttl to my **PC** on standard baudrate like 56700.  

I will type some char and send to **mega2560**. **mega2560** will route this with problematic baudrate to **Uno**. **Uno** will echo to **mega2560** and to PC. If char will be same then we can prove that USB to TTL is not able to support this baudrates.

Please let me know if this is good idea of we will need to debug this with oscilloscope.

explicite, 2020-11-08:
@Rahix looks like for mega2560 other than USART 0 required to lib update for board. There is type alias for
https://github.com/Rahix/avr-hal/blob/master/boards/arduino-mega2560/src/lib.rs#L25

Maybe we can try to  create USART file with all definitions in board?

Rahix, 2020-11-09:
Right, we can add a module to the `arduino-mega2560` crate with aliases for all the serial stuff:

```rust
pub mod usart {
    pub use avr_hal_generic::serial::*;
    pub type Usart0<IMODE> = ...;
    pub type Usart1<IMODE> = ...;
}
```

Rahix, 2020-11-10:
@explicite, just a quick heads up: I've pushed commits 493546530eb87d47ae1eee507e9309be590688d4 and 6b05d48b4e43a15b3610a0e852b261e61ab0cade which might affect you a bit.

explicite, 2020-11-10:
@Rahix, thanks, looks like I was able to apply this changes.
`14400` works fine with my test [mega2560](https://github.com/explicite/avr-hal/blob/0184ce3a15ca114bc78d23f8c8ab28b64f7096a8/boards/arduino-mega2560/examples/mega2560-serial.rs#L19-L41)
and [Uno](https://github.com/explicite/avr-hal/blob/0184ce3a15ca114bc78d23f8c8ab28b64f7096a8/boards/arduino-uno/examples/uno-serial.rs#L18-L31)
I was able to echo char with `14400`

Tomorrow will test two other bounds

Rahix, 2020-11-11:
>  Right now I'm using Leonardo as programmer and I need to re-upload bootloader each time. Can you advise which programmer can be a good for side projects? I have some experience with Atmel-ICE but not sure if is still good option. Thanks for advices.

Hmm, honestly I've just been using a Leonardo as ISP as well.  I've needed it so rarely thus far that this was not an issue at all.  Maybe I should ask why you need to re-upload the bootloader so often?  IMO that shouldn't ever be necessary and might hint at another bigger problem?

explicite, 2020-11-11:
Ok, I done testing for rest of the baudrates. Whole flow works fine.  I was able to get correct echo from Uno.

After changing `-cavrisp` to `-carduino` and with detached USART pins - only ICSP - I'm able to upload without re-uploading bootloader. Looks like I need housing and rewrite code to `avr-hal` and I can go with this solution for now ;) Thanks.

Is there any way to keep wiring and burn new code? Probably for this project is not a huge problem, but sometimes prototyping with Arduino can grow quickly.

Rahix, 2020-11-12:
> Ok, I done testing for rest of the baudrates. Whole flow works fine. I was able to get correct echo from Uno. Looks like I need to get better usbt to ttl. @Rahix, should we test this more?

Nice!  I think we still might to want to look at the signal on the wire with a logic analyzer/oscilloscope at some point but that is of much less priority now.  Thanks a lot for your work on this!

> Is there any way to keep wiring and burn new code? Probably for this project is not a huge problem, but sometimes prototyping with Arduino can grow quickly.

I'm not quite sure if I'm following.  What device/board are you working with here?

explicite, 2020-11-12:
In few days I will have logic analyzer, so will be able to debug this. 

According to programmers. I can see that attached wires to board can break bootloader when uploading code. Not sure why. I can see note in https://www.arduino.cc/en/Tutorial/BuiltInExamples/ArduinoISP according to this. 
Right now when I'm prototyping - e.g wiring some sensors - I need to detach all wires and then upload code.

Rahix, 2020-11-13:
Which board are you programming?  A bare AVR mcu or is this an Arduino board?  With the bootloader installed, the idea is that you can just upload a new program over USART, with no ISP required.  Does that not work?

explicite, 2020-11-14:
For mega2560 I can go with USART. No problem with uploading. Looks like there is a problem with Uno 
from `lsusb`
```
Bus 002 Device 014: ID 03eb:2fef Atmel Corp. atmega16u2 DFU bootloader
```
Cannot see any new `/dev`. Without any problems I'm able to upload via ISP. Will try to measure some resistors.

explicite, 2020-11-16:
I review all baudrates listed in ticket. I review logic state directly from `Uno` and after `USB to TTL`
1. All problematica baudrates works fine from `Uno` 
[image]
2. `USB to TTL` converts all listed rates to `9600`
3. I tested flow when from PC send with `14400` and `Uno` received correct data with `9600`

Looks like `USB to TTL` impl. nice magic under the hood. @Rahix should I test other scenarios?

Rahix, 2020-11-17:
> Looks like USB to TTL impl. nice magic under the hood.

Yeah, that's what I suspected ... Nice work digging all that up, I would say it is now safe to say that our code is not the problem here.  Thanks again!

> should I test other scenarios?

I think you've now covered every sensible test with what you did :)

closed 2020-11-17 by Rahix