Rahix/avr-hal #106: Question about Serial

mckayb, 2020-11-29:
Apologies if this is a dumb question, or if there is a better place to ask these sorts of questions, but I copied and pasted the arduino uno serial example and successfully flashed it to my arduino uno, but I'm unable to see any of the "Hello from Arduino!" or the "Got {}" messages.

I'm on a mac and ran 
```
screen /dev/tty.usbmodem144301 57600
```

But couldn't see any messages coming in.

Is there an equivalent to the Serial.print with the normal C/C++ variant that Arduino uses?
```
Serial.begin(57600);
Serial.print("Some value");
```
When I run the same screen with something like that, I can at least see values coming in.

Rahix, 2020-11-29:
Can you share more details what exactly you were doing?  How did you build the example?  How did you flash it?

mckayb, 2020-11-29:
All the code is here:
https://github.com/mckayb/rust-arduino/tree/project/love-o-meter/rust-arduino-03-love-o-meter

I copied it directly from:
https://github.com/Rahix/avr-hal/blob/master/boards/arduino-uno/examples/uno-serial.rs

To build it, I installed all of the avr tools, overrode rust to nightly, and just ran `cargo build` with the target set to the avr-atmega328p.json file listed in the code.

To flash it, I ran `avrdude -q -patmega328p -carduino -P/dev/tty.usbmodem144301 -D -Uflash:w:"target/avr-atmega328p/debug/rust-arduino-03-love-o-meter.elf"`

This was the output of both of those commands:
```
    Updating crates.io index
   Compiling rust-arduino-03-love-o-meter v0.1.0 (/Users/mckaybroderick/Code/rust-arduino/rust-arduino-03-love-o-meter)
    Finished dev [unoptimized + debuginfo] target(s) in 0.95s

avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "target/avr-atmega328p/debug/rust-arduino-03-love-o-meter.elf"
avrdude: input file target/avr-atmega328p/debug/rust-arduino-03-love-o-meter.elf auto detected as ELF
avrdude: writing flash (23882 bytes):
avrdude: 23882 bytes of flash written
avrdude: verifying flash memory against target/avr-atmega328p/debug/rust-arduino-03-love-o-meter.elf:
avrdude: load data flash data from input file target/avr-atmega328p/debug/rust-arduino-03-love-o-meter.elf:
avrdude: input file target/avr-atmega328p/debug/rust-arduino-03-love-o-meter.elf auto detected as ELF
avrdude: input file target/avr-atmega328p/debug/rust-arduino-03-love-o-meter.elf contains 23882 bytes
avrdude: reading on-chip flash data:
avrdude: verifying ...
avrdude: 23882 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

Then when I ran `screen /dev/tty.usbmodem144301 57600`
I just see a blank screen. When I type, nothing happens.

Thanks for taking a look at this. Let me know if there's something else you need.

Rahix, 2020-11-29:
Hmm, so you're building a debug build:

```text
    Finished dev [unoptimized + debuginfo] target(s) in 0.95s
```

and looking into your [`Cargo.toml`](https://github.com/mckayb/rust-arduino/blob/858db07a32a3f509d7516fd0181807d9344ad429/rust-arduino-03-love-o-meter/Cargo.toml#L18-L22), I see you've configured the compile profiles as follows:

```toml
[profile.dev]
panic = "abort"

[profile.release]
panic = "abort"
```

This will produce a huge bloated binary because absolutely no optimizations are done (no inlining, dead code removal, constant folding, etc.).  This means, it eats a lot of the available flash memory (23.3 KiB from your output above) but will also run very slowly.  I could see this massively throwing off the timing for the USART peripheral and thus nothing is working.  It could also run into a lot of other issues due to the missing optimizations.

I would recommend trying to configure the compiler to more aggresively optimize, in both debug and release builds.  In step 5 of [_Starting your own project_](https://github.com/Rahix/avr-hal#starting-your-own-project) in `avr-hal`'s README, the settings we recommend are these:

```toml
[profile.dev]
panic = "abort"
lto = true
opt-level = "s"

[profile.release]
panic = "abort"
codegen-units = 1
debug = true
lto = true
opt-level = "s"
```

I'd give those a try, maybe that is all that's needed to make it work :)

mckayb, 2020-11-29:
Ha, turns out, that was all it was! Thank you so much!

closed 2020-11-29 by mckayb