Rahix/avr-hal #108: split out atmega328pb

fneddy, 2020-12-01:
Hello,

I couldn't get the atmega328pb hal running. So i splinted it out in a separate sub-crate at `chips/atmega328pb-hal`. i think my problems came from setting the correct feature but then using them under a different name in the code. Nethertheless i think it is more clean to have a separate hal for each chip. If you think so to I can open a pull request, but if you want to keep it the way it is feel free to just close my issue :).

here are my changes: https://github.com/fneddy/avr-hal/commit/7e175153ce8a7607678df128d42a2eacb5d58963 it still need some polishing i think, but i could get a LED and UART working.

Rahix, 2020-12-02:
Hmm, just doing this should do the trick:

```toml
[dependencies.atmega328p-hal]
git = "https://github.com/Rahix/avr-hal"
rev = "<insert latest git-commit hash here>"
features = ["atmega328pb"]
```

(you probably also want the `"rt"` feature as well)

There is a reason we made those two a single crate and actually I intend to merge the HAL crates even more.  The current situation is becoming more and more unmaintainable as all changes need to be reflected across all crates.  This huge amount of code-duplication is becoming a time-sink and preventing fast iteration on the APIs we expose.  Other HAL crates in the embedded-rust ecosystem seem to be doing the same, so we want to follow that route. See #94 for some more info.

jkristell, 2020-12-02:
I have an example using the atmega328pb-hal here: https://github.com/jkristell/rustyremote

Been a couple of weeks since I worked on it, but it seems to still compile.

fneddy, 2020-12-02:
Thank you. I'll have a look. I close this issue through it is not needed. :)

closed 2020-12-02 by fneddy