Rahix/avr-hal #113: use of undeclared crate or module `ufmt`

MLackner, 2020-12-09:
Hi!

Throughout the examples 
```rust
ufmt::writeln!(&mut serial, ...);
```
is used. If I do that I'm not able to compile and I get the following error:
```rust
48 |     ufmt::uwriteln!(&mut serial, "Hello from Arduino!\r").void_unwrap();
   |     ^^^^ use of undeclared crate or module `ufmt`
```

I tried to add `ufmt` under `[dependencies]` in the Cargo.toml as `ufmt = "0.1.0"`. Doing so gives me access to the ufmt crate with `use ufmt;`. The code then compiles (although `rust-analyzer` in VSCode complains that it ``could not resolve macro `ufmt::uwriteln` ``).

Is this how it is intended to work? Would be great if someone could help me with that! :)

Rahix, 2020-12-09:
Hey!

Yes, you need to add `ufmt` as a dependency to make this work.  I guess the documentation should be more clear on that ...

Why `rust-analyzier` is failing, I have no idea.  I'm also using it and did not see any such problems so far.  Can you get a more verbose output of what it is complaining about?

nbigaouette, 2021-02-14:
I have the same issue regarding rust-analyzer. Here's my repo: https://github.com/nbigaouette/robot-car/tree/343b89bc397dc99761de583d744ebe8d47dfd135/rust

I can cargo build and flash to my arduino uno (thanks for avr-hal!!) but rust-analyzer cannot find `ufmt`:
[image]

My guess is that rust-analyzer is confused with the target?

nbigaouette, 2021-02-14:
I've reported the rust-analyzer issue upstream: https://github.com/rust-analyzer/rust-analyzer/issues/7674

nbigaouette, 2021-02-14:
Turns out that `ufmt` uses proc-macro-hack which rust-analyzer does not supports... 🤷

Rahix, 2021-02-21:
Okay, as this is not really something related to `avr-hal` and there is an upstream report now, I'll close this issue here.

closed 2021-02-21 by Rahix