Rahix/avr-hal #119: unneeded argument in pin `into_` functions

fneddy, 2020-12-14:
Hi,

all pin `into_*` functions have a `ddr: &D`(data direction register) argument.

https://github.com/Rahix/avr-hal/blob/d17a441a667dd65e5a9be75b0e71d1beadf93e84/avr-hal-generic/src/port.rs#L510-L562

this is actually not needed, because the `DDR` name comes from the macro expansion (`$reg_ddr`).
is it ok to remove this arguments?

Rahix, 2020-12-14:
> this is actually not needed, because the DDR name comes from the macro expansion ($reg_ddr).

Whether the argument is used here or not is actually a secondary matter: It exists as a safety guarantee to enforce synchronous access.  I.e. `ddr` acts as a token which represents the DDR register and code should only perform DDR accesses while holding the token.  The actual access then happens via an unsafe direct access to the register location solely because that's easier to express in the current macro architecture than trying to unpack the `AsDDR` type (at least that was the point when I wrote it, not sure if that is still true?).

Now, whether this kind of enforcement is actually needed is a different matter: It is a question about if the accesses we need to perform can be done atomically.  Only if this is the case, can we drop this strict synchronous access enforcement.

The answer to that question looks promising:  As we essentially do single-bit sets and clears the optimizations which I explored in [this gitter comment](https://gitter.im/avr-rust/Lobby?at=5fbe9c0bfe857d71dc94e3d3) do apply.  For the newer AVR MCUs, the IO-cell even provides dedicated registers to perform those operations atomically.  So I'd say with the rework in #114 we can drop the DDR parameter from the conversion functions.

koutoftimer, 2021-02-16:
@Rahix add gitter to the main Readme pls

Rahix, 2021-02-16:
Right, might make sense add a "community"/contact section there.  So far people have always found their way over there themselves though ^^

Rahix, 2021-02-16:
Regarding this issue, actually: Commit 9046c06794c5 ("generic: port: Add new port implementation") which is currently on the `next` branch resolves this; the new port API does not require explicit DDR passing anymore.

closed 2021-02-16 by Rahix