Rahix/avr-hal #121: How to add a handle to an interupt?

tronta, 2020-12-26:
I'm a embedded newbie and I like to implement a Servo library for the Arduino in Rust. Thanks, first of all, for your library.
Sorry, if this is not the appropriate channel to post this question.

So the idea is to implement a robot control in Rust. When studying the Servo.cpp code it looks like the handle_interrupts is triggered continuously by a timer: https://github.com/arduino/Steelwonder/blob/167b594c3d09417df6257c5f7cb4a6c3194785b6/Firmware%20(arduino%20sketch)/Marlin/Servo.cpp#L77
which then continuously writes the values to the pins.
Is my observation right? What would be the best way with Rust? I would need it for The Mega processor.

Thanks a lot!

lights0123, 2020-12-26:
See https://github.com/Rahix/avr-hal/issues/75#issuecomment-702675340

tronta, 2020-12-27:
Cool. Thanks for the link. I'll have a look.

closed 2020-12-27 by tronta