Rahix/avr-hal #122: Ensure that you are using an AVR target! You may need to change directories or pass a --target flag to cargo

ThisIsRex, 2020-12-30:
I followed a tutorial on the main page, but `cargo build` throws this error. I don't found any information on the page thiss error points to. But if i use this command:
`cargo build -Z build-std=core --target avr-atmega328p.json --release`
all will be allright
How can I build using `cargo build` only? It seems cargo doesn't see a build target in cargo.toml

cargo.toml
`[package]
name = "arduino-test"
version = "0.1.0"
authors = ["Admin"]
edition = "2018"

[build]
target = "avr-atmega328p.json"

[unstable]
build-std = ["core"]

[dependencies]
panic-halt = "0.2.0"
ufmt = "0.1.0"
nb = "1.0.0"

[dependencies.arduino-uno]
git = "https://github.com/Rahix/avr-hal"
rev = "d17a441a667dd65e5a9be75b0e71d1beadf93e84"

[profile.dev]
panic = "abort"
lto = true
opt-level = "s"

[profile.release]
panic = "abort"
codegen-units = 1
debug = true
lto = true
opt-level = "s"`

lights0123, 2020-12-30:
Copy the .cargo/config.toml file present in every board crate here into the root folder of your crate

ThisIsRex, 2020-12-31:
> 
> 
> Copy the .cargo/config.toml file present in every board crate here into the root folder of your crate

thanks!

closed 2020-12-31 by ThisIsRex