Rahix/avr-hal #123: how to create a new board crate outside of avr-hal?

MarkuBu, 2021-01-06:
Hi, I want to create a new board, but outside of the avr-hal crate. How can I do this?

Regards

Rahix, 2021-01-06:
I think the easiest way is to copy an existing board crate directory and adjust all the necessary bits in `Cargo.toml`.  For example:

```toml
[dependencies.atmega328p-hal]
git = "https://github.com/Rahix/avr-hal.git"
rev = "<latest git commit hash from avr-hal>"
features = ["atmega328p"]

[dependencies.avr-hal-generic]
git = "https://github.com/Rahix/avr-hal.git"
rev = "<latest git commit hash from avr-hal>"
```

Rahix, 2021-01-06:
You'll also want to include the target json file for your MCU, e.g. `avr-specs/avr-atmega328p.json`.  You'll then need to adjust the path to it in `.cargo/config.toml`.

MarkuBu, 2021-01-06:
Thanks for the quick response. I don't know how to adjust the path to the target JSON file. At least I don't know how to set the correct path in Windows

Rahix, 2021-01-06:
Let's say your new board crate is in the `my-board/` directory.  I.e. there is a `my-board/Cargo.toml` for the crate.  Copy `avr-specs/avr-atmega328p.json` into this directory, so `my-board/avr-atmega328p.json`.  Then put the following in `my-board/.cargo/config.toml`:

```toml
[build]
target = "avr-atmega328p.json"

[unstable]
build-std = ["core"]
```

MarkuBu, 2021-01-06:
Missed the [unstable] part, but still doesn't work. Here is my cargo.toml

```
[package]
name = "dt_rs"
version = "0.1.0"
authors = ["Markus Bu"]
edition = "2018"

[build]
target = "avr-atmega2560.json"

[unstable]
build-std = ["core"]

[features]
default = ["rt"]
rt = ["atmega2560-hal/rt"]

[dependencies.atmega2560-hal]
git = "https://github.com/Rahix/avr-hal.git"
rev = "d17a441"
features = ["atmega2560"]

[dependencies.avr-hal-generic]
git = "https://github.com/Rahix/avr-hal.git"
rev = "d17a441"

[dev-dependencies]
panic-halt = "0.2.0"
nb = "0.1.2"
ufmt = "0.1.0"

```

Now I get the error message

```
error: failed to select a version for `atmega2560-hal`.
    ... required by package `dt_rs v0.1.0 (C:\Users\Markus\dev\dt_rs)`
versions that meet the requirements `*` are: 0.1.0

the package `dt_rs` depends on `atmega2560-hal`, with features: `atmega2560` but `atmega2560-hal` does not have these features.

failed to select a version for `atmega2560-hal` which could resolve this conflict
```

Rahix, 2021-01-06:
No, you put `.cargo/config.toml` contents into the same file as `Cargo.toml`.  Those are two entirely separate things:

- `Cargo.toml` is the crate "manifest"
- `.cargo/config.toml` is configuration for cargo, e.g. things you can also select with command-line flags.

You'll have to remove the `[build]` and `[unstable]` sections from `Cargo.toml` and put them into a _separate_ file named `config.toml` which is in a subfolder called `.cargo/`.

Additionally, you need to remove the

```toml
features = ["atmega2560"]
```

because that's not needed for `ATmega2560` which you're using.

MarkuBu, 2021-01-06:
*Facepalm*

Thanks, now I can build with --debug, but with --release I get this error
```
error: could not compile `core`

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 C:\Users\Markus\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=a8a8392fd8dfb55b -C extra-filename=-a8a8392fd8dfb55b --out-dir C:\Users\Markus\dev\dt_rs\target\avr-atmega2560\release\deps --target \\?\C:\Users\Markus\dev\dt_rs\avr-atmega2560.json -Z force-unstable-if-unmarked -L dependency=C:\Users\Markus\dev\dt_rs\target\avr-atmega2560\release\deps -L dependency=C:\Users\Markus\dev\dt_rs\target\release\deps --cap-lints allow` (exit code: 0xc0000005, STATUS_ACCESS_VIOLATION)
warning: build failed, waiting for other jobs to finish...
error: build failed
```

Maybe this is related to this Issue

https://github.com/avr-rust/template-bin/issues/1

But the solution doesn't work, at least on this Windows machine at work. On Linux at home it worked.

But the other error is  (signal: 11, SIGSEGV: invalid memory reference), this error is (exit code: 0xc0000005, STATUS_ACCESS_VIOLATION)

Rahix, 2021-01-06:
Hm, I can't really help you with that unfortunately ... This sounds a lot like some LLVM bug you're hitting.

For a start, I suggest adding the profiles from step 5 of the [Starting your own project](https://github.com/Rahix/avr-hal#starting-your-own-project) guide to your `Cargo.toml`.  Those have worked well for me so far.

MarkuBu, 2021-01-06:
Thanks for your help so far. I will try it at home at my Linux PC.

I really need Linux at work

koutoftimer, 2021-02-16:
@MarkuBu Virtual Box or VM Ware Workstation is out there for you to help

stappersg, 2021-02-19:
@MarkuBu another path to success is telling your manager that you have successful compiles on Linux  and discuss working tools.

Rahix, 2021-02-21:
@MarkuBu, I'm going to close this issue for now.  If you need further help with this, just leave a comment then we can reopen it.

closed 2021-02-21 by Rahix