Rahix/avr-hal #124: LLVM ERROR: Not supported instr

sobolevn, 2021-01-08:
I am using this example: https://github.com/Rahix/avr-hal/tree/d17a441a667dd65e5a9be75b0e71d1beadf93e84/boards/arduino-uno

I have also executed `rustup component add rust-src`

And it fails to compile with this error:

```
» cargo +nightly build --example uno-blink
   Compiling compiler_builtins v0.1.39
   Compiling nb v1.0.0
   Compiling cfg-if v0.1.10
   Compiling bare-metal v0.2.5
   Compiling void v1.0.2
   Compiling vcell v0.1.3
   Compiling ufmt-write v0.1.0 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
   Compiling panic-halt v0.2.0
   Compiling nb v0.1.3
   Compiling avr-device v0.2.3
   Compiling ufmt v0.1.0 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
   Compiling embedded-hal v0.2.4
LLVM ERROR: Not supported instr: <MCInst 258 <MCOperand Reg:1> <MCOperand Imm:15> <MCOperand Reg:40>>
error: could not compile `compiler_builtins`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

My versions:

- rust

```
» rustup --version
rustup 1.23.1 (2020-12-03)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.51.0-nightly (c8915eebe 2021-01-07)`
```

- avr
```
» avr-gcc --version
avr-gcc (Homebrew AVR GCC 9.3.0) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

- And macos 10.14

lights0123, 2021-01-08:
Try `--release`?

sobolevn, 2021-01-08:
It is a bit different, but still the same 🙂 

```
» cargo +nightly build --example uno-blink --release 
   Compiling compiler_builtins v0.1.39
   Compiling core v0.0.0 (/Users/sobolev/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.58
   Compiling semver-parser v0.7.0
   Compiling proc-macro-hack v0.5.19
   Compiling paste v1.0.4
   Compiling semver v0.9.0
   Compiling rustc_version v0.2.3
   Compiling quote v1.0.8
   Compiling bare-metal v0.2.5
   Compiling rustc-std-workspace-core v1.99.0 (/Users/sobolev/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling avr-device-macros v0.2.3
   Compiling ufmt-macros v0.1.1 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
   Compiling nb v1.0.0
   Compiling ufmt-write v0.1.0 (https://github.com/Rahix/ufmt.git?rev=12225dc1678e42fecb0e8635bf80f501e24817d9#12225dc1)
   Compiling void v1.0.2
   Compiling vcell v0.1.3
   Compiling cfg-if v0.1.10
   Compiling panic-halt v0.2.0
   Compiling nb v0.1.3
   Compiling avr-device v0.2.3
   Compiling embedded-hal v0.2.4
LLVM ERROR: Not supported instr: <MCInst 258 <MCOperand Reg:1> <MCOperand Imm:15> <MCOperand Reg:40>>
error: could not compile `compiler_builtins`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

sobolevn, 2021-01-08:
I have made a reproduction:

```Dockerfile
FROM rustlang/rust:nightly-slim

RUN apt-get update \
  && apt-get install --no-install-recommends -y \
    avr-libc \
    binutils-avr \
    gcc-avr \
  && apt-get purge -y --auto-remove -o APT::AutoRemove::RecommendsImportant=false \
  && apt-get clean -y && rm -rf /var/lib/apt/lists/*

COPY . /code

WORKDIR /code/boards/arduino-uno

RUN rustup component add rust-src
RUN cargo +nightly build --example uno-blink

```

Steps:
0. Add this `Dockerfile` in the root of the project
0,5. Ignore or delete any `target/` dirs
1. Run `docker build --tag=ard .`

This is the output:

```
Step 6/6 : RUN cargo +nightly build --example uno-blink
 ---> Running in fde0189b88d8
    Updating git repository `https://github.com/Rahix/ufmt.git`
    Updating crates.io index
 Downloading crates ...
  Downloaded nb v0.1.3
  Downloaded paste v1.0.4
  Downloaded rustc_version v0.2.3
  Downloaded semver-parser v0.7.0
  Downloaded quote v1.0.8
  Downloaded proc-macro2 v1.0.24
  Downloaded proc-macro-hack v0.5.19
  Downloaded vcell v0.1.3
  Downloaded void v1.0.2
  Downloaded cfg-if v0.1.10
  Downloaded unicode-xid v0.2.1
  Downloaded syn v1.0.58
  Downloaded avr-device-macros v0.2.3
  Downloaded avr-device v0.2.3
  Downloaded semver v0.9.0
  Downloaded panic-halt v0.2.0
  Downloaded nb v1.0.0
  Downloaded embedded-hal v0.2.4
  Downloaded bare-metal v0.2.5
 Downloading crates ...
  Downloaded cc v1.0.60
  Downloaded compiler_builtins v0.1.39
  Downloaded getopts v0.2.21
  Downloaded hashbrown v0.9.0
  Downloaded addr2line v0.14.0
  Downloaded gimli v0.23.0
  Downloaded miniz_oxide v0.4.0
  Downloaded adler v0.2.3
  Downloaded object v0.22.0
  Downloaded unicode-width v0.1.8
  Downloaded rustc-demangle v0.1.18
  Downloaded libc v0.2.79
   Compiling compiler_builtins v0.1.39
   Compiling core v0.0.0 (/usr/local/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.58
   Compiling semver-parser v0.7.0
   Compiling proc-macro-hack v0.5.19
   Compiling paste v1.0.4
   Compiling semver v0.9.0
   Compiling rustc_version v0.2.3
   Compiling quote v1.0.8
   Compiling bare-metal v0.2.5
   Compiling rustc-std-workspace-core v1.99.0 (/usr/local/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
LLVM ERROR: Not supported instr: <MCInst 258 <MCOperand Reg:1> <MCOperand Imm:15> <MCOperand Reg:40>>
error: could not compile `compiler_builtins`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
The command '/bin/sh -c cargo +nightly build --example uno-blink' returned a non-zero code: 101
```

lights0123, 2021-01-08:
Regression in `nightly-2021-01-08, rust version 1.51.0-nightly (c8915eebe 2021-01-07)`, should probably file a [compiler-builtins](https://github.com/rust-lang/compiler-builtins/) bug. [Diff](https://github.com/rust-lang/rust/compare/c2de47a9a..c8915eebe), probably because the last 4 months of changes in compiler-builtins were used.

Rahix, 2021-01-25:
We have identified the problem, let's see how it will be solved.  In the meantime, for anyone stumbling upon this issue:  Just use an older toolchain like `nightly-2021-01-07` for now.  With rustup, installing and using it will be as easy as:

```bash
rustup toolchain install nightly-2021-01-07
# ...
cargo +nightly-2021-01-07 build
```

Urhengulas, 2021-01-25:
Thank you for the workaround!
Is there a way to support you in fixing it?

Ewpratten, 2021-03-01:
For anyone that needs this information:

Rust `nightly-2020-11-20` is the last version that was built that both doesn't have this issue, and had a successful `rls` build.

So, if you are (for example) looking for a release where you can work with `avr-hal` and have VSCode give you useful autocomplete and highlighting, `2020-11-20` is the version to use.

Sources: [`toolsuite build info`](https://raw.githubusercontent.com/rust-lang-nursery/rust-toolstate/master/history/linux.tsv) and [`rustdatehash`](https://gist.github.com/krdln/11cdbcf2d3742d40ed7b)

sobolevn, 2021-03-01:
One more solution to `rls` problem (`2021-01-07` does not have `rls` component): `rls` can be replaced with https://github.com/rust-analyzer/rust-analyzer

mike-kfed, 2021-03-10:
I found it's easiest to create a file called `rust-toolchain` in your projects directory with following content

```toml
[toolchain]
channel = "nightly-2021-01-07"
components = ["rust-src"]
```
afterwards you can simply run `cargo build` and all is fine :)

Ewpratten, 2021-03-10:
For my own future reference: [these](https://rust-lang.github.io/rustup/overrides.html#the-toolchain-file) are the docs for `rust-toolchain` files.

Urhengulas, 2021-03-10:
> I found it's easiest to create a file called `rust-toolchain` in your projects directory with following content
> 
> ```toml
> [toolchain]
> channel = "nightly-2021-01-07"
> components = ["rust-src"]
> ```
> 
> afterwards you can simply run `cargo build` and all is fine :)

This is actually super helpful. Thanks for the tip!

lf-, 2021-03-15:
This is https://github.com/rust-lang/rust/issues/82104 and https://github.com/rust-lang/compiler-builtins/issues/400

waynevanson, 2021-06-14:
> Rust `nightly-2020-11-20` is the last version that was built that both doesn't have this issue, and had a successful `rls` build.

Thank you! I was looking for this!

djdisodo, 2021-07-09:
gcc backend will be soon supported for rust
so can we use them instead?

stappersg, 2021-07-09:
On Thu, Jul 08, 2021 at 09:39:44PM -0700, sodo wrote:
> gcc backend will be soon supported for rust
> so can we use them instead?

Soon we can find out

stappersg, 2021-07-09:
Pardon my way too whity response.

Yes please, keep sharing possible solutions.
Do notify when GCC Rust arrives.

ssmgcode, 2021-08-05:
Hi everyone. Same problem here, I just `cargo generate --git https://github.com/Rahix/avr-hal-template.git` and then `cargo build`.
The output:
```bash
$ cargo build
   Compiling compiler_builtins v0.1.49
   Compiling core v0.0.0 (/home/ssmg/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.28
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.74
   Compiling semver-parser v0.7.0
   Compiling proc-macro-hack v0.5.19
   Compiling paste v1.0.5
   Compiling semver v0.9.0
   Compiling rustc_version v0.2.3
   Compiling quote v1.0.9
   Compiling bare-metal v0.2.5
   Compiling rustc-std-workspace-core v1.99.0 (/home/ssmg/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling nb v1.0.0
   Compiling cfg-if v0.1.10
   Compiling void v1.0.2
   Compiling ufmt-write v0.1.0
   Compiling vcell v0.1.3
LLVM ERROR: Not supported instr: <MCInst 296 <MCOperand Reg:1> <MCOperand Imm:15> <MCOperand Reg:39>>
error: could not compile `compiler_builtins`
warning: build failed, waiting for other jobs to finish...
error: build failed
```

I've tried `rustup toolchain install nightly-2021-01-07` and in my rust-toonchail.toml:
```toml
[toolchain]
channel = "nightly-2021-01-07"
components = [ "rust-src" ]
```
and then `cargo build` I got:
```bash
   Compiling compiler_builtins v0.1.49
   Compiling nb v1.0.0
   Compiling bare-metal v0.2.5
   Compiling ufmt-write v0.1.0
   Compiling vcell v0.1.3
   Compiling void v1.0.2
   Compiling cfg-if v0.1.10
   Compiling cfg-if v1.0.0
   Compiling panic-halt v0.2.0
   Compiling ufmt v0.1.0
   Compiling nb v0.1.3
   Compiling avr-device v0.3.1
   Compiling embedded-hal v0.2.6
LLVM ERROR: Not supported instr: <MCInst 296 <MCOperand Reg:1> <MCOperand Imm:13> <MCOperand Reg:40>>
error: could not compile `compiler_builtins`
warning: build failed, waiting for other jobs to finish...
error: build failed
```
and with `cargo +nightly-2021-01-07 build` I got:
```bash
error: "/home/ssmg/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
        rustup component add rust-src
```
but I make `rustup component add rust-src` and I obtain the same output.

Thank you for your help and time.

lights0123, 2021-08-05:
@ssmgcode you need to add the rust-src component for _the particular toolchain_. I believe you can use the same +«toolchain-version» trick you can use with other Rust commands.

Rahix, 2021-08-05:
@ssmgcode, it seems your rustup is not picking up `rust-toolchain.toml`?  What version of rustup do you have?  Maybe you need to do a `rustup self update`?

ssmgcode, 2021-08-06:
> @ssmgcode you need to add the rust-src component for _the particular toolchain_. I believe you can use the same +«toolchain-version» trick you can use with other Rust commands.

Thank you! I made `rustup +nightly-2021-01-07 component add rust-src`, deleted the project folder and create it again from template. Then I just `cargo build` and it worked nicely!

ssmgcode, 2021-08-06:
> @ssmgcode, it seems your rustup is not picking up `rust-toolchain.toml`? What version of rustup do you have? Maybe you need to do a `rustup self update`?

Yes, it seemed, it was a really weird thing. My rustup version is `rustup 1.24.3 (ce5817a94 2021-05-31)`, I think it's a recent version because I started using Rust and installed it a month ago, sort of.

Thank you for your time guys, this makes me happy.

djdisodo, 2021-09-29:
[libgccjit-codegen has been merged (1.57.0)](https://github.com/rust-lang/rust/pull/87260)
i hope this can be used instead of llvm, solve this problem

Patryk27, 2022-05-08:
Heads-up: https://github.com/rust-lang/rust/pull/96845 🙂

Rahix, 2022-05-09:
This is huge, thank you so much for your efforts here @Patryk27!

PR #268 (2022-05-10): Upgrade to a recent rust compiler nightly
Through the work of @Patryk27, nightly can compile for AVR again as of today :)  Upgrade the nightly version referenced in the toolchain file to opt people into the new compiler if they are using rustup.

Finally closes #124.  Thanks again @Patryk27!
closed 2022-05-10 by Rahix