Rahix/avr-hal #131: Possible to build without the --release flag?

kkevlar, 2021-02-09:
Hi all,

Every time I try to build and Arduino Uno project without the --release flag, my project is unable to link with the panic handler.

```
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-Os" "-mmcu=atmega328p" "-L" "/home/kevlar/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "/home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/busted-48e39b52f28b50c4.arduino_uno-aa6bd09237895391.arduino_uno.bng1ym0s-cgu.0.rcgu.o.rcgu.o" "-o" "/home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/busted-48e39b52f28b50c4.elf" "-Wl,--gc-sections" "-L" "/home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps" "-L" "/home/kevlar/Documents/rust-morse/uno/target/debug/deps" "-L" "/home/kevlar/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,--start-group" "-Wl,--end-group" "-Wl,-Bstatic" "/home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/libcompiler_builtins-22ecd261b63a219e.rlib" "-Wl,-Bdynamic" "-Wl,--gc-sections"
  = note: /home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/libcompiler_builtins-22ecd261b63a219e.rlib(compiler_builtins-22ecd261b63a219e.compiler_builtins.2xqjeibm-cgu.6.rcgu.o): In function `compiler_builtins::int::mul::Mul::mul::hf2ef2c06cb9c88ef':
          /home/kevlar/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/mul.rs:(.text._ZN17compiler_builtins3int3mul3Mul3mul17hf2ef2c06cb9c88efE+0x29e): undefined reference to `core::panicking::panic::h80cbf9e666e18772'
          /home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/libcompiler_builtins-22ecd261b63a219e.rlib(compiler_builtins-22ecd261b63a219e.compiler_builtins.2xqjeibm-cgu.6.rcgu.o): In function `compiler_builtins::int::mul::Mulo::mulo::h1a93e217f9703dd5':
          /home/kevlar/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/mul.rs:(.text._ZN17compiler_builtins3int3mul4Mulo4mulo17h1a93e217f9703dd5E+0x3ac): undefined reference to `core::panicking::panic::h80cbf9e666e18772'
          /home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/libcompiler_builtins-22ecd261b63a219e.rlib(compiler_builtins-22ecd261b63a219e.compiler_builtins.2xqjeibm-cgu.7.rcgu.o): In function `compiler_builtins::int::shift::Ashl::ashl::h1eb3228f4f805c9e':
          /home/kevlar/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/shift.rs:(.text._ZN17compiler_builtins3int5shift4Ashl4ashl17h1eb3228f4f805c9eE+0x150): undefined reference to `core::panicking::panic::h80cbf9e666e18772'
          /home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/libcompiler_builtins-22ecd261b63a219e.rlib(compiler_builtins-22ecd261b63a219e.compiler_builtins.2xqjeibm-cgu.7.rcgu.o): In function `compiler_builtins::int::shift::Ashl::ashl::h828bb5286455f201':
          /home/kevlar/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/shift.rs:(.text._ZN17compiler_builtins3int5shift4Ashl4ashl17h828bb5286455f201E+0x198): undefined reference to `core::panicking::panic::h80cbf9e666e18772'
          /home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/libcompiler_builtins-22ecd261b63a219e.rlib(compiler_builtins-22ecd261b63a219e.compiler_builtins.2xqjeibm-cgu.7.rcgu.o): In function `compiler_builtins::int::shift::Lshr::lshr::h08c3922bd0713bd0':
          /home/kevlar/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/shift.rs:(.text._ZN17compiler_builtins3int5shift4Lshr4lshr17h08c3922bd0713bd0E+0x1f8): undefined reference to `core::panicking::panic::h80cbf9e666e18772'
          /home/kevlar/Documents/rust-morse/uno/target/avr-atmega328p/debug/deps/libcompiler_builtins-22ecd261b63a219e.rlib(compiler_builtins-22ecd261b63a219e.compiler_builtins.2xqjeibm-cgu.1.rcgu.o):/home/kevlar/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/specialized_div_rem/delegate.rs:(.text._ZN17compiler_builtins3int19specialized_div_rem11u64_div_rem17hc28005c952b8fec9E+0x125c): more undefined references to `core::panicking::panic::h80cbf9e666e18772' follow
          collect2: error: ld returned 1 exit status
          

error: aborting due to previous error; 13 warnings emitted

error: could not compile `busted`
```

Is this expected behavior? Thank you so much for your help.

Rahix, 2021-02-09:
Yeah, this is a known behavior, unfortunately.  The problem is that in debug builds, `overflow-checks` are enabled.  Due to a problem in cargo's `build-std`, they are not only enabled for your code, but also for the `compiler-builtins` crate.  This means, the compiler will insert overflow checks into the intrinsics defined in `compiler-builtins`, which in turn reference the panic handler symbol.  Now due to the way the Rust standard library works, this is not possible (binary object linking order).  That's why you get the errors.

In a non-`build-std` build, e.g. with a precompiled `libcore` and `compiler-builtins` this wouldn't be a problem.

There is a workaround although it isn't really pretty.  To quote a [previous comment of mine](https://github.com/Rahix/avr-hal/issues/125#issuecomment-771578390):

> that's a known issue: rust-lang/compiler-builtins#347
>
> There is a dirty workaround to add the following to your `Cargo.toml`:
>
> ```toml
> [profile.dev.package.compiler_builtins]
> overflow-checks = false
> ```
>
> `cargo` will warn about this but it'll still work fine... Alternatively compile in release mode where `overflow-checks` are always disabled.

One other option would be to disable `overflow-checks` entirely in debug mode as well (i.e. also for all other crates including your code), but that of course is a dangerous decision.  You'd do this by adding

```toml
[profile.dev]
overflow-checks = false
```

kkevlar, 2021-02-11:
Thank you for your help. I'll use this for now.

Rahix, 2021-02-12:
Glad I could help!  Let's keep this issue open for visibility until the problem is fixed.