Rahix/avr-hal #137: Ensure that you are using an AVR target! You may need to change directories or pass a --target flag to cargo

Kurren123, 2021-02-11:
I'm on windows

I'm getting this error when I follow online tutorials. My cargo.toml:

```toml
[package]
name = "rust-arduino-blink"
version = "0.1.0"
authors = [""]
edition = "2018"

[dependencies]
panic-halt = "0.2.0"

[dependencies.arduino-uno]
git = "https://github.com/Rahix/avr-hal"

[build]
target = "avr-atmega328p.json"

[unstable]
build-std = ["core"]
```

And my `main.rs`:
```rust
#![no_std]
#![no_main]

extern crate panic_halt;

use arduino_uno::prelude::*;

#[arduino_uno::entry]
fn main() -> ! {
    let dp = arduino_uno::Peripherals::take().unwrap();

    unimplemented!()
}
```

Any help would be much appreciated!

Rahix, 2021-02-12:
You need to read the instructions more carefully ;)

This part: 

```toml
[build]
target = "avr-atmega328p.json"

[unstable]
build-std = ["core"]
```

does not belong into `Cargo.toml`, but into a separate file named `.cargo/config.toml`.

Also, you need more directives in your `Cargo.toml` to configure the build profile.  Take another look at the [Starting your own project](https://github.com/Rahix/avr-hal#starting-your-own-project) guide to see the details.

Kurren123, 2021-02-12:
@Rahix  Thanks for your help! I've now created the `Cargo.toml` and `.cargo/config.toml`, repository [here](https://github.com/Kurren123/rust-arduino-blink). However I'm now getting this error when I do `rustup override set nightly` and `cargo build --verbose`:

```
error: could not compile `compiler_builtins`

Caused by:
  process didn't exit successfully: `rustc --crate-name compiler_builtins "C:\Users\Kurren N\.cargo\registry\src\github.com-1ecc6299db9ec823\compiler_builtins-0.1.39\src\lib.rs" --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -C linker-plugin-lto -C debuginfo=2 -C debug-assertions=on --cfg "feature=\"compiler-builtins\"" --cfg "feature=\"core\"" --cfg "feature=\"default\"" --cfg "feature=\"rustc-dep-of-std\"" -C metadata=c49ed016198a86a1 -C extra-filename=-c49ed016198a86a1 --out-dir C:\dev\Arduino\rust-arduino-blink\target\avr-atmega328p\debug\deps --target \\?\C:\dev\Arduino\rust-arduino-blink\avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=C:\dev\Arduino\rust-arduino-blink\target\avr-atmega328p\debug\deps -L dependency=C:\dev\Arduino\rust-arduino-blink\target\debug\deps --extern core=C:\dev\Arduino\rust-arduino-blink\target\avr-atmega328p\debug\deps\librustc_std_workspace_core-494a22ba4d488457.rmeta --cap-lints allow --cfg "feature=\"unstable\""` (exit code: 101)
```

Any suggestions?

Rahix, 2021-02-12:
Please try using `nightly-2021-01-07` instead (#124):

```bash
rustup toolchain install nightly-2021-01-07
rustup override set nightly-2021-01-07
```

Also, you should probably pin the version of avr-hal in your dependencies like this:

```toml
[dependencies.arduino-uno]
git = "https://github.com/rahix/avr-hal"
rev = "b1aedf824fcb59078f8022d4464b2835ae4bd83a"
```

Kurren123, 2021-02-12:
Thanks, this now worked! Only issue is that my IDE is showing errors as rust language server is not available on that nightly date, and it's hard to find the most recent nightly build before the 8th jan that includes rls.

Thanks again for your help!

closed 2021-02-12 by Kurren123