Rahix/avr-hal #146: Do you need more examples for arduino uno?

crclark96, 2021-02-26:
I'm in an embedded software class and decided to do the assignments in Rust with the help of this crate, so I have a bunch of minimal examples doing things like reading from a DHT11 temp sensor, using a timer interrupt (which I actually sourced from your blog, so you probably have this), and I'm currently working an example with external interrupts.

I'm happy to work with you to get the source code in a state that's consistent with your other examples, but if you feel there are enough examples I'll just put them on my page (since some of them have to access registers directly since the interrupt abstraction layer isn't quite ready yet).

Let me know what you think --

stappersg, 2021-02-26:
> Let me know what you think

I think _more experiments like #147_.    Yes, I'm saying  _patches welcome_.

Additional info:  I'm not a project member.

Rahix, 2021-02-27:
In my eyes, more examples is always better - they help people understand how a project is intended to be used in ways that documentation just can't explain.  So I'll take everything I can get :)

We're currently doing quite big changes to `avr-hal`, see #130.  So I think it would be best to wait until that refactor is done and then rework the examples for the new API.

crclark96, 2021-03-01:
Great - I'll keep my eye on that issue

As an aside, some of these have ugly interrupt initialization akin to the `millis()` example -- is there a plan for implementing an interrupt API? I didn't see it in #130

Rahix, 2021-03-01:
I assume by "ugly interrupt initialization" you mean that you have to write the registers directly?

Well, I do intend to add an API similar to [the one for USART](https://rahix.github.io/avr-hal/atmega328p_hal/usart/struct.Usart.html#method.listen) to all other peripheral drivers as well at some point.  Or are you looking for something else?

crclark96, 2021-03-01:
"Ugly" is the wrong word -- unabstracted? 

Either way, that API looks similar to what I was asking about, and I saw discussion about a timer API so that should cover everything I've implemented by hand, thanks

victor-wei126, 2022-11-12:
Is there an example to use the LCD with Rust crate? Arduino language has a simple header file that can instantiate lcd, and python has many libraries. Was wondering if this crate has something similar or we have to do it manually?

Rahix, 2022-11-14:
What LCD are you trying to drive, @victor-wei126?  The idea in the embedded Rust community is to have driver crates for each LCD controller and those can be connected to each microcontroller HAL, including `avr-hal` here.  So find a fitting LCD driver crate and try attaching it to the peripheral types provided from `avr-hal`.

Just as a guess, maybe something like this is what you need? https://crates.io/crates/hd44780-driver

Otherwise, check this list of driver crates: https://github.com/rust-embedded/awesome-embedded-rust

If you need help, let us know.

Rahix, 2022-11-14:
(And in line with the original issue topic here, if someone wants to contribute an LCD example, please do!)

victor-wei126, 2022-11-29:
Those crates look promising @Rahix. I will check them out. Thanks for the help.