Rahix/avr-hal #148: Modulo operator error

davystrong, 2021-02-27:
Not sure if this is the correct place to report this or even if this is a known issue, but the modulo operator seems to have a problem. It works as expected when used with powers of two, but returns 1 when used with other numbers. This doesn't happen when running on my laptop. The following code demonstrates the issue on the Arduino Uno:
```
#![no_std]
#![no_main]
#![feature(abi_avr_interrupt)]

extern crate panic_halt;

use arduino_uno::prelude::*;

const PWM_PERIOD: i32 = 200;

#[arduino_uno::entry]
fn main() -> ! {
    let dp = arduino_uno::Peripherals::take().unwrap();

    let mut pins = arduino_uno::Pins::new(dp.PORTB, dp.PORTC, dp.PORTD);

    let mut serial = arduino_uno::Serial::new(
        dp.USART0,
        pins.d0,
        pins.d1.into_output(&mut pins.ddr),
        57600.into_baudrate(),
    );

    let mut val: i32 = 0;

    loop {
        ufmt::uwriteln!(&mut serial, "{}\r", (val % PWM_PERIOD)).void_unwrap();
        val += PWM_PERIOD / 2;
        arduino_uno::delay_ms(100);
    }
}
```
Let me know if this should be reported somewhere else.

Rahix, 2021-02-27:
Hey!

This is probably related to https://github.com/rust-lang/rust/issues/82242 ...

davystrong, 2021-02-27:
Great, thanks. I'll watch that issue for updates. And thanks for your work implementing this package!

closed 2021-02-27 by davystrong