Rahix/avr-hal #161: Official Support for Arduino Mirco

jnnksdev, 2021-03-09:
There seems to be missing support for Arduino Micro. It seems to be similar to Sparkfun Pro Micro, so most of the implementation might be c&p. The blink example already works without a problem.

I would be happy to help, please let me know, how i can assist.

Arduino Micro in Store: [store.arduino.cc/arduino-micro](https://store.arduino.cc/arduino-micro)

Rahix, 2021-03-09:
There's currently a refactoring going on in avr-hal, see #130.  If you want, you can take a look at some of the commits adding support for new boards over there and do something similar for Arduino Micro.  For example, commits 5304fd0d923c ("arduino-hal: Add support for Arduino Leonardo") and 8f6928cbfd26 ("Add example crate for Arduino Leonardo") add support for Arduino Leonardo which also uses the ATmega32U4.

Be sure to do this against the `next` branch of this repository, where the refactor is ongoing.  The `master` branch is **not** up-to-date with that.

If you have any questions, let me know :)

michelepagot, 2021-09-07:
I'm interested too. I also have a Genuino MICRO on my desk to test. I managed to update the avr-hal-template to start with ...