Rahix/avr-hal #173: uno-runner for Windows Subsystom for Linux

Septias, 2021-04-08:
This is a post on how to connect serial devices to WSL: https://www.scivision.dev/usb-tty-windows-subsystem-for-linux/

On Linux, the Arduino doesn't show up as `/dev/ttyACM0` but as for example `/dev/ttyS3` when it's connected to Port 3.

This would perhaps conclude that changing this: `SERIAL_PORT="/dev/ttyACM0"` in uno-runner.sh to `SERIAL_PORT="/dev/ttyS3"` would do the trick but that's not the case I think because I get ` Inappropriate ioctl for device` instead. 

How can I make this work?

Rahix, 2021-04-08:
The runner script is actually a thing of the past.  I've been working on a proper tool for this, see https://blog.rahix.de/ravedude/ :)  It currently lives in the `next` branch of this repository, waiting to be merged eventually (#130).

I don't use windows so I have no clue how well it works over there.  Maybe you can install `ravedude` and give it a try?

Septias, 2021-04-08:
well thanks for the really fast response and the website of ravedude is as nice as its name, but when I run `cargo install ravedude` I get this mess:

```
   Compiling heck v0.3.2
error[E0433]: failed to resolve: could not find `addr_of` in `ptr`
   --> /home/septias/.cargo/registry/src/github.com-1ecc6299db9ec823/anyhow-1.0.40/src/error.rs:606:14
    |
606 |         ptr::addr_of!((*unerased.as_ptr())._object) as *mut E,
    |              ^^^^^^^ could not find `addr_of` in `ptr`

error[E0433]: failed to resolve: could not find `addr_of` in `ptr`
   --> /home/septias/.cargo/registry/src/github.com-1ecc6299db9ec823/anyhow-1.0.40/src/error.rs:647:22
    |
647 |                 ptr::addr_of!((*unerased.as_ptr())._object) as *mut E,
    |                      ^^^^^^^ could not find `addr_of` in `ptr`

error: failed to run custom build command for `libudev-sys v0.1.4`

Caused by:
  process didn't exit successfully: `/tmp/cargo-installtHrT5U/release/build/libudev-sys-56c2ffec939741e5/build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "Failed to run `\"pkg-config\" \"--libs\" \"--cflags\" \"libudev\"`: No such file or directory (os error 2)"', /home/septias/.cargo/registry/src/github.com-1ecc6299db9ec823/libudev-sys-0.1.4/build.rs:38:41
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0433`.
error: failed to compile `ravedude v0.1.2`, intermediate artifacts can be found at `/tmp/cargo-installtHrT5U`

Caused by:
  build failed
```

Septias, 2021-04-08:
the latter part 
```
Caused by:
  process didn't exit successfully: `/tmp/cargo-installtHrT5U/release/build/libudev-sys-56c2ffec939741e5/build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "Failed to run `\"pkg-config\" \"--libs\" \"--cflags\" \"libudev\"`: No such file or directory (os error 2)"', /home/septias/.cargo/registry/src/github.com-1ecc6299db9ec823/libudev-sys-0.1.4/build.rs:38:41
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: aborting due to 2 previous errors
```
can be eliminated with `Ubuntu/Debian: apt install libudev-dev pkg-config` which is on crates.io but not on your website, but the compile errors still remain

Rahix, 2021-04-08:
Looks like you have an older compiler version, maybe?

Septias, 2021-04-08:
Well the readme says it needs to be capped to work

Septias, 2021-04-08:
Yeah, I only installed WSL for this project and since the version should be capped according to README I only had that version installed. I installed newest one and it now works - at least installing.
Running it with this command: 
```toml
[target.'cfg(target_arch = "avr")']
runner = "ravedude uno --open-console --baudrate 57600 -P/dev/ttyS3"
```
still produces `Inappropriate ioctl for device` :/

Rahix, 2021-04-08:
Hm, can you try using it outside WSL?  Not sure but you might be lucky and it could just work...

Septias, 2021-04-08:
I don't have a Linux machine :/ Might try it on windows tho...

Rahix, 2021-04-08:
Ah, I was referring to trying it on windows directly, sorry!

drmorr0, 2021-04-09:
Are you using WSL2 or 1?  If you're using 2, I believe it can't yet communicate with the serial ports.  The workaround I came up with was to install avrdude _in Windows_ and then call that executable from WSL, because that can read the serial ports.  I'm now using a programmer and Atmel studio, so I can't comment on how ravedude works, unfortunately :\

Septias, 2021-04-09:
Okay, that sounds like a reasonable approach, can you send me your avr-command? I tried a little bit but I just can't get it right. Since we're having the same problem it should be possible to use yours, right?

Septias, 2021-04-09:
I just moved to WSL 1 and that works with ravedude.

Not exactly what I wanted but works for now so I'll close it

closed 2021-04-09 by Septias

H4xX0r1337, 2021-04-10:
@Septias @Rahix I was able to get it working by manually calling the script with the elf file as parameter using **git sh on windows**  
This edit is needed in the script. Check the COM port in the Arduino IDE. 
`if [ $OS = "Linux" ]; then
    SERIAL_PORT="/dev/ttyACM0"
elif [ $OS = "Mac" ]; then
    SERIAL_PORT="/dev/cu.usbmodem146201"
else
    echo "unsupported OS, things might not work" >&2
    SERIAL_PORT="COM3"
fi`

Rahix, 2021-04-11:
@H4xX0r1337, I see.  But as I said those scripts are to be removed anyway, please try [`ravedude`](https://crates.io/crates/ravedude)  instead.  Ideally it should also work natively on windows, if it doesn't it would be good to find out why.

H4xX0r1337, 2021-04-11:
@Rahix When installing ravedude on Windows I get the same issue as Septias (`error[E0433]: failed to resolve: could not find `addr_of` in `ptr``)  
And I don't want to install cargo on WSL because that would take 500mb of space again and would make stuff way more complicated for me.

Rahix, 2021-04-11:
@H4xX0r1337 this error is caused by using an older version of the rust compiler.  Please update to 1.51.0 and try again.  WSL should not be needed, the goal is to make it work without that entirely.

Septias, 2021-04-12:
Where should I post issues with ravedude? 

I tried it on Windows 10 19042.867 but it doesn't work: 
```
 Programming target\avr-atmega328p\debug\arduino.elf => COM4
avrdude: invalid file format 'e' in update specifier
avrdude: error parsing update operation 'flash:w:target\avr-atmega328p\debug\arduino.elf:e'
Error: avrdude failed

error: process didn't exit successfully: `ravedude uno --open-console --baudrate 57600 target\avr-atmega328p\debug\arduino.elf` (exit code: 1)
```
It complains about :e specifier being wrong and that's exactly what I observed when I tried to flash manually on windows.

stappersg, 2021-04-12:
FWIW in https://github.com/Rahix/avr-hal/issues/176#issuecomment-818008234 we did learn that a recent `avrdude` gives better result.