Rahix/avr-hal #176: Ravedude not working on Windows (10)

Septias, 2021-04-12:
Running rave dude like this on Windows 10 
```toml
[target.'cfg(target_arch = "avr")']
runner = "ravedude uno --open-console --baudrate 57600"
```
produces this error: 
```rust
Programming target\avr-atmega328p\debug\arduino.elf => COM4
avrdude: invalid file format 'e' in update specifier
avrdude: error parsing update operation 'flash:w:target\avr-atmega328p\debug\arduino.elf:e'
Error: avrdude failed
error: process didn't exit successfully: `ravedude uno --open-console --baudrate 57600 target\avr-atmega328p\debug\arduino.elf` (exit code: 1)
```

It complains about `:e` specifier being wrong and that's exactly what I observed when I tried to flash manually on windows.

Rahix, 2021-04-12:
Thanks!  Can you quickly check which version of `avrdude` you have installed?

Septias, 2021-04-12:
$ ravedude -V  
`ravedude v0.1.2 (no git)`

Rahix, 2021-04-12:
right, but I was looking for the **avrdude** version, not _ravedude_ ;)  Try

```
$ avrdude -?
```

the version will be shown at the very end of the help output.

Septias, 2021-04-12:
Oof still version 5.10... with 6.3 it works...

Septias, 2021-04-12:
I'm still experiencing "performance issues" with the serial console though... 
It should normally output a message with `ufmt` every 250ms. On my Linux WSL with a screen  it runs  really smooth, but the console created by ravedude always outputs the messages in batches of like 3-5 messages instead of one-by-one.
When I open the connection manually: 
```powershell
function read-com {
    $port= new-Object System.IO.Ports.SerialPort COM3,57600,None,8,one
    $port.Open()
    do {
        $line = $port.ReadLine()
        Write-Host $line # Do stuff here
    }
    while ($port.IsOpen)
}
```
I get a steady flow.

stappersg, 2021-04-12:
On Mon, Apr 12, 2021 at 11:27:38AM -0700, Sebastian Klähn wrote:
> experiencing "performance issues" with the serial console

Please in a fresh issue ...

Rahix, 2021-04-13:
> Please in a fresh issue ...

No need to.

---

> Oof still version 5.10... with 6.3 it works...

I see.  The question is, if lot's of people still use 5.10 we should support it as well.  Where did you get this older version?

---

> I'm still experiencing "performance issues" with the serial console though...

Hm, this looks like a buffering issue.  But we're actually flushing stdout whenever a new message arrives:

https://github.com/Rahix/avr-hal/blob/9e3848f13b7ef3b03617545c959ad476e454a76a/ravedude/src/console.rs#L20-L21

Maybe you can play around with that code a bit to check what is going on?  I don't have a windows on hand so I cannot test this...

You can test just the serial console (without flashing a binary) by calling:

```text
$ ravedude uno -cb 57600
```

(without a path to a binary).

Rahix, 2021-04-13:
(or test directly from the `ravedude/` directory with `cargo run -- -cb 57600`)

Rahix, 2021-04-13:
Maybe try removing this line and see if it changes anything:

https://github.com/Rahix/avr-hal/blob/9e3848f13b7ef3b03617545c959ad476e454a76a/ravedude/src/console.rs#L7

michelepagot, 2021-09-07:
> Oof still version 5.10... with 6.3 it works...

hi all, I'm in the same situation and I fall in the same mistake. Is it possible to cite in the documentation that at least avrdude => 6.3 has to be used or something like "tested with avrdude 6.3" ?

stappersg, 2021-09-07:
On Tue, Sep 07, 2021 at 09:07:21AM -0700, Michele wrote:
> that at least avrdude => 6.3 has to be used or something like "tested with avrdude 6.3" ?

Thank you for https://github.com/Rahix/avr-hal/pull/218.patch

Rahix, 2024-05-04:
I think over time all the problems from this issue got resolved.  So I will close it for now, even though I don't know for sure what the status is.  If anyone is still experiencing issues, we can reopen the topic again.

closed 2024-05-04 by Rahix