Rahix/avr-hal #183: error: linking with `avr-gcc` failed... cannot find crtatmega328p.o

MarkuBu, 2021-05-19:
Hi, I want to try again to write code with avr-hal, but again I have problems to get set it up.

I want to compile the uno-blink example, but I get this error message

```
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-mmcu=atmega328p" "-Wl,--as-needed" "-L" "/home/markus/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "/home/markus/dev/avr-hal/target/avr-atmega328p/debug/examples/uno_blink-e99c14a2ae3aaf44.panic_halt-0008bfb58d1acc80.panic_halt.eegxnupy-cgu.0.rcgu.o.rcgu.o" "-o" "/home/markus/dev/avr-hal/target/avr-atmega328p/debug/examples/uno_blink-e99c14a2ae3aaf44.elf" "-Wl,--gc-sections" "-L" "/home/markus/dev/avr-hal/target/avr-atmega328p/debug/deps" "-L" "/home/markus/dev/avr-hal/target/debug/deps" "-L" "/home/markus/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,--start-group" "-Wl,--end-group" "-Wl,-Bstatic" "/home/markus/dev/avr-hal/target/avr-atmega328p/debug/deps/libcompiler_builtins-c59fe7dbb9f0d488.rlib" "-Wl,-Bdynamic" "-lgcc"
  = note: /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: cannot find crtatmega328p.o: No such file or directory
          collect2: error: ld returned 1 exit status
          

error: aborting due to previous error

error: could not compile `arduino-uno`

```

I'm running Lubuntu 21.04. I installed the gcc-avr and build-essential packages. Do I miss something?

Regards

closed 2021-05-19 by MarkuBu

stappersg, 2021-05-19:
On Wed, May 19, 2021 at 10:51:46AM -0700, MarkuBu wrote:
>  ... 
> Do I miss something?

What was missing?

As in:  "How could we improve our documentation?"

Rahix, 2021-05-19:
@MarkuBu, yeah, once the new version gets finalized, I'll overhaul the docs and include more details on the necessary requisites to be installed.  Ideally we want to get rid on the dependency on avr-libc at some point but right now I do not have enough time to work on that...