Rahix/avr-hal #191: Solved but for other beginners: invalid record at line ...

enaut, 2021-07-09:
Hey,

thanks for that crate! I'm trying to get my arduino-nano to work. During flashing it reports: invalid record at line and a line number.

The output when I try to flash:
```
❯ avrdude -C/etc/avrdude/avrdude.conf -v -patmega168 -carduino -P/dev/ttyUSB0 -b19200 -D -Uflash:w:target/avr-atmega168/release/rust-arduino-blink.elf:i

avrdude: Version 6.3, compiled on Jan 26 2021 at 00:00:00
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude/avrdude.conf"
         User configuration file is "/home/dietrich/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega168
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no        512    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9406 (probably m168)
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file "target/avr-atmega168/release/rust-arduino-blink.elf"
avrdude: invalid record at line 51 of "target/avr-atmega168/release/rust-arduino-blink.elf"
avrdude: read from file 'target/avr-atmega168/release/rust-arduino-blink.elf' failed

avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

enaut, 2021-07-09:
### Solution

The problem was that the `.elf` file cannot be flashed directly, but rather needs to be "extracted" to a hex file using:

```
$ avr-objcopy -O ihex target/avr-atmega168/release/rust-arduino-blink.elf arduino-blink.hex
$ avrdude -C/etc/avrdude/avrdude.conf -v -patmega168 -carduino -P/dev/ttyUSB0 -b19200 -D -Uflash:w:arduino-blink.hex:i
```

closed 2021-07-09 by enaut

stappersg, 2021-07-09:
Please consider to create a patch against our documentation.

Rahix, 2021-07-09:
@enaut, you seem to have an old version of `avrdude`.  Maybe consider upgrading?  Also, your Arduino Nano has an `ATmega328P`, not an `ATmega168` so something is off with your commandline...

That said, I suggest you give the [`ravedude`](https://crates.io/crates/ravedude) tool a shot.  It takes care of calling avrdude for you with the correct parameters, integrates with cargo nicely, and has a serial console if you need it.

enaut, 2021-07-09:
@Rahix thanks for the reply - I do have avrdude 6.3 that seems to be recent - its from the fedora repositories? I do have an arduino nano with atmega168 and one with atmega328p did I mix something?

ravedude looks awesome!

Rahix, 2021-07-09:
> I do have avrdude 6.3 that seems to be recent

Ah, wait, I was so used to seeing these problems due to old avrdude that I didn't look close enough.  You're commandline is wrong, the `:i` at the end needs to be `:e` for elf files.

> I do have an arduino nano with atmega168 and one with atmega328p did I mix something?

I wasn't aware of any nano with ATmega168... Do you have any online information about that?  I only know of the ATmega328P version.

enaut, 2021-07-10:
Oh so thanks `:e` is indeed way easier than manually converting.

It's not an actual arduino but a clone:
Many of the ebay clones are with the smaller chip for even cheaper nanos... just search for `arduino nano 168` on ebay...

I didn't know about different chips either... and bought a small one instead of a 328p but the program I want to fit is too big. Hence I bought another with the bigger chip.