Rahix/avr-hal #194: Question: is postible to write a spefic value to analog port?

luukvankooten, 2021-07-14:
With the Arduino HAL, it is possible to write values between 0 and 255 with the function `analogWrite()`. How to do this with this library?

Rahix, 2021-07-15:
Hi,

first of all, it is important to note that `analogWrite()` does not actually produce an analog signal, but instead uses [PWM](https://en.wikipedia.org/wiki/Pulse-width_modulation) under the hood.  PWM signals are generated by the MCUs timer peripherals and then "routed" to one of the supported pins.

Now, we had support for this at some point, but as we recently reworked pretty much all of `avr-hal` (see #130), this is unfortunately still missing from the new version.  This means that you'll have to set up the timer for PWM manually.

As an example, here is code to output a PWM signal on pin D9 of an Arduino Uno.  The frequency of the signal is 1 kHz and we're varying the duty-cycle from 0% to 100%.  (Disclaimer: I can't test this code right now, but I'm pretty sure this is correct)

```rust
#![no_std]
#![no_main]

use arduino_hal::prelude::*;
use panic_halt as _;

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_uno::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);
    let mut serial = arduino_hal::default_serial!(dp, pins, 57600);

    ufmt::uwriteln!(&mut serial, "Hello from Arduino!").void_unwrap();

    // This is important because otherwise the signal will not actually be
    // visible on the pin.
    pins.d9.into_output();

    let tc1 = dp.TC1;
    tc1.tccr1a.write(|w| w.wgm1().bits(0b01).com1a().match_clear());
    tc1.tccr1b.write(|w| w.wgm1().bits(0b01).cs1().prescale_64());

    loop {
        for duty in 0u8..=255u8 {
            ufmt::uwriteln!(&mut serial, "Duty: {}", duty).void_unwrap();
            tc1.ocr1a.write(|w| unsafe { w.bits(duty as u16) });
            arduino_hal::delay_ms(20);
        }
    }
}
```

You can of course do the same thing for all the other PWM-capable pins, but the timer and registers to use will be different.  The datasheet for the MCU contains all this information, but if you're unsure you can of course always ask me as well.

luukvankooten, 2021-07-15:
Thank you, but to which pin is this writing?

stappersg, 2021-07-15:
@luukvankooten consider to express your question in other words.

Expressing _Oops, I do now see "As an example, here is code to output a PWM signal on pin D9 of an Arduino Uno." in the text above the source_  is also fine.

Rahix, 2021-07-15:
@stappersg, what's your point?  The OP asked a clear question and got a (hopefully) comprehensive answer.  What's missing?

stappersg, 2021-07-15:
Missing  is the comment that got me this email:
```text
Date: Thu, 15 Jul 2021 03:39:41 -0700                                                                                               
From: Luuk van Kooten <notifications@github.com>                                                                                    
To: Rahix/avr-hal <avr-hal@noreply.github.com>                                                                                      
Cc: Subscribed <subscribed@noreply.github.com>                                                                                      
Subject: Re: [Rahix/avr-hal] Question: is postible to write a spefic value to analog port? (#194)                                   
                                                                                                                                    
Thank you, but to which pin is this writing?                                                                                        
                                                                                                                                    
--                                                                                                                                  
You are receiving this because you are subscribed to this thread.                                                                   
Reply to this email directly or view it on GitHub:                                                                                  
https://github.com/Rahix/avr-hal/issues/194#issuecomment-880589489    
```

To prove my point I will remove my comment from a hour ago.

No hard feelings.

Rahix, 2021-07-15:
Ah, now it makes more sense - my GH notifications aren't routed over mail so I never got to see that comment (off-topic, but why does GH even allow removing comments without any trace at all?).  Thanks for clearing it up!

stappersg, 2021-07-15:
> Thanks for clearing it up!

`:-)`

luukvankooten, 2021-07-16:
After long debugging hours (I'm new to this with rust, data sheets and microcontrollers). The line containing `    pins.d9.into_output();` must be under the lines tc1.tccr1a and  tc1.tccr1b. I think it has something to do with registration, not sure.

closed 2021-07-18 by luukvankooten

MarkuBu, 2021-07-18:
analogWrite and similar functions are higher abstraction functions. In my opinion it doesn't make sense to add them in a HAL for a controller.

If you have a controller that has an actual analog output like a DAC you need different functions. 

These kind of functions only make sense in a higher abstraction layer that is independent of the actual hardware

Rahix, 2021-07-18:
@MarkuBu, while I agree that an `analogWrite()` function makes no sense here, I do think that we need to add an abstraction for the PWM outputs which can serve as a functionally equivalent replacement.  Just like the one we had in old `avr-hal`, see <https://github.com/Rahix/avr-hal/blob/ce52c90d3950cd81090627e3ad8494a15e4c8f66/boards/arduino-uno/examples/uno-pwm.rs>.

miam-miam100, 2021-12-03:
Hi I am trying to get the PWM working on an Arduino uno on pins 5 and 6 to emulate the `analogWrite()` function but it doesn't work at all. Would it be possible for someone to help me?
```rust
#![no_std]
#![no_main]

use arduino_hal::prelude::*;
use panic_halt as _;

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);
    let mut serial = arduino_hal::default_serial!(dp, pins, 57600);

    ufmt::uwriteln!(&mut serial, "Hello from Arduino!").void_unwrap();

    // Using timer 0 as pin 5+6 on Arduino Uno use it instead of timer 1.
    let tc0 = dp.TC0;
    tc0.tccr0a
        .write(|w| w.wgm0().bits(0b01).com0a().match_clear());
    tc0.tccr0b
        .write(|w| w.wgm02().bit(true).cs0().prescale_64());

    // Setting output after as this is what luukvankooten suggested
    pins.d5.into_output();
    pins.d6.into_output();

    loop {
        for duty in 0u8..=255u8 {
            ufmt::uwriteln!(&mut serial, "Duty: {}", duty).void_unwrap();
            tc0.ocr0a.write(|w| unsafe { w.bits(duty) });
            tc0.ocr0b.write(|w| unsafe { w.bits(duty) });
            arduino_hal::delay_ms(20);
        }
    }
}
```

Rahix, 2021-12-03:
@miam-miam100, please open a new issue for this.

emendir, 2025-02-23:
I figured out what seems to me a decent way to abstract away this low-level PWM functionality into something that allows you to use PWM like this:
```rs
// Initialise board IO
let (mut pin_map, mut pwm) = setup_pins();

// Pin Output Examples:

// PWM output 
pwm.write(&pin_map.pin_5, 100);

// PWM output given pin number, like Arduino's analogWrite
pwm.write_to_pin_number(3, 150);
```
To learn how to do this, see
https://github.com/emendir/arduino-rust-pwm-example

Rahix, 2025-02-25:
@emendir, are you aware of the `simple_pwm` module in `avr-hal`?  See [example](https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/src/bin/uno-simple-pwm.rs) and [documentation](https://rahix.github.io/avr-hal/arduino_hal/simple_pwm/index.html).

emendir, 2025-03-01:
> [@emendir](https://github.com/emendir), are you aware of the `simple_pwm` module in `avr-hal`? See [example](https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/src/bin/uno-simple-pwm.rs) and [documentation](https://rahix.github.io/avr-hal/arduino_hal/simple_pwm/index.html).

Oh, that makes things a lot easier of course!