Rahix/avr-hal #206: Ravedude needs nightly to compile

wt, 2021-08-01:
Without nightly, this error occurs:
```
✗ cargo build           
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names -Clink-arg=-fuse-ld=lld -Zshare-generics=y --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit status: 1)
  --- stderr
  error: the option `Z` is only accepted on the nightly compiler
```

Rahix, 2021-08-01:
I cannot reproduce this.  What compiler version are you using?  What revision of `ravedude` are you building?

For reference, building  2996b4a7885d ("generic: port: Implement `embedded-hal` OutputPin trait") with

```text
❯ cargo --version && rustc --version
cargo 1.53.0 (4369396ce 2021-04-27)
rustc 1.53.0 (53cb7b09b 2021-06-17)
```

works perfectly fine for me.

wt, 2021-08-01:
I am getting this on main at commit c728d7b8b4f01794795634578c6f5ec37b8a7480. Here's my rustup data from that directory:

```
➜  ravedude git:(main) rustup show active-toolchain -v    
stable-x86_64-unknown-linux-gnu (overridden by '/home/wt/projects/rust_arduino/avr-hal/ravedude/rust-toolchain.toml')
rustc 1.54.0 (a178d0322 2021-07-26)
```

The cargo version is also coming from stable:
```
 ➜  ravedude git:(main) rustup which cargo
/home/wt/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/bin/cargo
```

Rahix, 2021-08-01:
Still cannot reproduce, with the exact same versions which your are using...  Did you make sure to `cargo clean` before rebuilding?  Do you have any odd settings in your `~/.cargo/config`?  Or `RUSTFLAGS` set in the environment?  I find it very odd that a `-Z` option would randomly get added to the compiler invocation...

Also, `--crate-name ___` looks odd.  What's that about?

wt, 2021-08-01:
You were right. I did have some cruft in my .cargo/config from when I was playing with bevy. I nuked it and it's fine. Thanks for the help.

closed 2021-08-01 by wt