Rahix/avr-hal #211: Using i2c to PWM 16 servo x 12bit board, how would I send a signal to a servo?

joshuajbouw, 2021-08-09:
Hey there, this is a general question. I also understand that this is a fairly low-level library which I am completely fine with. However, I am struggling to find good information on exactly what the data packet should be to be able to control a servo on board 0x40. If you can point me in the right direction, I'll be more than grateful.

Rahix, 2021-08-09:
You need to provide more information, I cannot help you from this.  What do you mean by "Using i2c to PWM 16 servo x 12bit board"?  Do you have an I2C PWM port-expander which you're using to control a servo?  Please provide a description of the circuit (even better: schematic), and exact part numbers for the components involved.  Otherwise there is no way I can know what you need...

joshuajbouw, 2021-08-09:
Sure.

# Hardware
- Arduino Uno
- [Adafruit 16-Channel 12-bit PWM/Servo Driver](https://www.adafruit.com/product/815)
- [Tower Pro MG995](https://www.electronicoscaldas.com/datasheet/MG995_Tower-Pro.pdf)

# Connections
- Arduino Uno connected to PWM.
    - Arduino `A4` connected to PWM `SDA`
    - Arduino `A5` connected to PWM `SCL`
    - Arduino `5V` connected to PWM `VCC`
    - Arduino `GND` connected to PWM `GND`
    
 I had followed the `i2c` [example](https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/src/bin/uno-i2cdetect.rs) for Arduino Uno and the detection is successful. However, I only see that I can send bytes to a board at an address. In my case 0x40. What does that data need to look like?

Rahix, 2021-08-09:
Thanks.  It looks like that board of yours uses a PCA9685 "16-channel, 12-bit PWM Fm+ I2C-bus LED controller" ([datasheet](https://cdn-shop.adafruit.com/datasheets/PCA9685.pdf)).  You'll need to correctly write to its registers to archieve your desired goal.  There are three routes you can take here:

1. You're in luck and there already is an embedded-rust driver for this chip: [`pwm-pca9685`](https://crates.io/crates/pwm-pca9685). You might be able to just hook that up to the i2c driver from avr-hal and then use its API to control the generated PWM signal.  Here is an example I found, which might help: [`pca9685-servos-f3.rs`](https://github.com/eldruin/driver-examples/blob/master/stm32f3-discovery/examples/pca9685-servos-f3.rs).
2. You can look at the Adafruit driver what it does ([`Adafruit_PWMServoDriver.cpp`](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/Adafruit_PWMServoDriver.cpp)) and try replicating the same in Rust.  It will most likely be very similar to the crate mentioned above, though.
3. You can take the information from the datasheet linked above and try piecing together your own servo driver, if all else fails.  This is probably not that straight forward though...

joshuajbouw, 2021-08-10:
Works flawlessly. Thank you.

Rahix, 2021-08-10:
That's great to hear!  Would you mind sharing some sample code here, for future readers, to demonstrate how you got it working?  Even better, maybe submit an example as a PR, e.g. as `examples/arduino-uno/src/bin/uno-16chan-servo-driver.rs`

joshuajbouw, 2021-08-11:
Sure, can do so once I am through work for the day.