Rahix/avr-hal #213: Using `alloc` with the RAM?

joshuajbouw, 2021-08-14:
Hey there! I tried toying with this but I'm unsure if its because it isn't implemented yet. There is RAM on the Arduino that can be used. For example 2,048 bytes for the Uno. How can I access this through the library?

Rahix, 2021-08-15:
You are already using the SRAM because your program stack and data is located there.  The concept of a HEAP does not really exist at this point (or on plain embedded platforms in general).  Depending on what you want to do, that's not even something you need - many embedded applications work quite fine without one and in doing so circumvent a large number of problems.

As an alternative, you can push things you'd usually allocate on the heap into a `static`.  That's equivalent to a global variable/`static` in C.  As an example:

```rust
static mut VAR: Option<TYPE> = None;

// initialize it
unsafe { VAR = Some(value); }

// get an &mut reference to it (will panic if not initialized)
let var_ref = unsafe { VAR.as_mut().unwrap() };
```

Of course this needs unsafe for a reason - there are a few invariants you must uphold here.

---

On the other hand, it is possible to make `alloc` work - albeit I should warn you that there are some pitfalls.  Especially because SRAM is so limited on AVR, you must be very careful that A) your stack isn't overflowing and B) memory fragmentation is not becoming an issue.

To make it work, you must first define a "heap" by defining a large buffer as a `static`.  Then you can use this buffer with an allocator implementation like [`linked-list-allocator`](https://crates.io/crates/linked_list_allocator).  I have never actually tried this on an AVR target yet, but I don't see a fundamental problem which should prevent it from working.

ahmtcn123, 2021-11-16:
Is there a example for using vectors or alloc?

Rahix, 2021-11-17:
I don't think so.  But as stated above, just set up [`linked-list-allocator`](https://crates.io/crates/linked_list_allocator) with a large `static mut` buffer as your heap and you should be good to go.

ahmtcn123, 2021-11-19:
Can you give me a example? I couldn't find any examples or docs available

Rahix, 2021-11-21:
Don't have an example available, but really just follow the `linked-list-allocator` README by using a large `static mut` as storage:

```rust
static mut HEAP: [u8; 1024] = [0u8; 1024];

#[global_allocator]
static ALLOCATOR: LockedHeap = LockedHeap::empty();

fn main() {
    unsafe {
        let heap_start = HEAP.as_ptr() as usize;
        let heap_size = HEAP.len(); // or core::mem::size_of(HEAP);
        ALLOCATOR.lock().init(heap_start, heap_size);
    }
}
```

(I have never tried this on AVR though, so I cannot guarantee that it works there)