Rahix/avr-hal #214: Static access to Serial interface on arduino nano

Rechenmaschine, 2021-08-17:
Hey there, lately I've been trying to implement my own Messaging protocol for the arduino nano, however I have been having some problems realising this because I cannot find the explicit type of the `Usart`. 

I need the explicit type definition of the `Usart`, because I need to either pass it as a parameter, or have static access to it. I am not able to call `Usart::new()` multiple times either, because the values I pass are moved. 

Does anyone have a solution or a workaround on hand for this? 

Thanks in advance!

Rahix, 2021-08-17:
Hi,

you can store your `Usart` in a global static and then define a macro-wrapper for accessing it.  Roughly like this (untested):

```rust
#![no_std]
#![no_main]

use arduino_hal::prelude::*;
use panic_halt as _;

pub mod serial {
    use avr_device::interrupt::Mutex;
    use core::cell::RefCell;

    pub type Usart = arduino_hal::hal::usart::Usart0<arduino_hal::DefaultClock>;
    pub static GLOBAL_SERIAL: Mutex<RefCell<Option<Usart>>> = Mutex::new(RefCell::new(None));

    pub fn init(serial: Usart) {
        avr_device::interrupt::free(|cs| {
            GLOBAL_SERIAL.borrow(&cs).replace(Some(serial));
        })
    }

    #[macro_export]
    macro_rules! serial_println {
        ($($arg:tt)*) => {
            ::avr_device::interrupt::free(|cs| {
                if let Some(serial) = &mut *crate::serial::GLOBAL_SERIAL.borrow(&cs).borrow_mut() {
                    ::ufmt::uwriteln!(serial, $($arg)*)
                } else {
                    Ok(())
                }
            })
        }
    }
}

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);
    let serial = arduino_hal::default_serial!(dp, pins, 57600);

    serial::init(serial);

    serial_println!("Hello from Arduino!\r").void_unwrap();

    loop { }
}
```

Rahix, 2021-08-17:
> I need the explicit type definition of the Usart, because I need to either pass it as a parameter

Ah, yeah so that's the

```rust
    pub type Usart = arduino_hal::hal::usart::Usart0<arduino_hal::DefaultClock>;
```

line from above...

Rechenmaschine, 2021-08-17:
Thanks! Works as expected.

closed 2021-08-17 by Rechenmaschine