Rahix/avr-hal #216: Proper way to exit Ravedude serial monitor?

lord-ne, 2021-08-25:
What is the proper way to exit the Ravedude serial monitor? Exiting with `CTRL+C` gives `error: process didn't exit successfully: 'ravedude uno --open-console --baudrate 9600 target\avr-atmega328p\release\arduino_test.elf' (exit code: 0xc000013a, STATUS_CONTROL_C_EXIT)`. It still works fine, but being shown an error message is just slightly bad user experience. 

The reason why I want to exit without closing the terminal is because I want to use the same terminal to compile/run the code again.

stappersg, 2021-08-25:
On Wed, Aug 25, 2021 at 07:53:19AM -0700, lord-ne wrote:
> Exiting with `CTRL+C` gives `error: process didn't exit successfully

And at which platform that?

Please state clearly what we are talking about.
At least leave the idea "We have all the same configuration".

 
> The reason why I want to exit without closing the terminal
> is because I want to use the same terminal to compile/run the code again.

Yes, the wish is a reasonable wish.

Regards
Geert Stappers

Rahix, 2021-09-08:
Okay, as I do not have a windows machine to try this, maybe someone else can do this for me:  Try using the [`ctrlc`](https://crates.io/crates/ctrlc) crate to catch a `^C` and make the handler just call [`std::process::exit()`](https://doc.rust-lang.org/std/process/fn.exit.html).  If this works, I think it would be a good workaround for now.

It is just a workaround because `std::process::exit` does not properly clean up before exiting.  Ideally pressing `^C` should make the main function exit cleanly.  But we can't archieve that easily because there is (AFAIK) no portable way to non-blockingly read from STDIN...  So the callstack of `main()` is, for the most time, blocked in the `read(STDIN)` call and thus cannot be notified of the `^C` happening.

PR #225 (2021-11-01): ravedude: Cleanly terminate on CTRL+C
Install a CTRL+C handler to terminate with exit code 0 instead of the default handler which exits with a failure.

@lord-ne, can you give this a test-run on your windows system to see if the error message is gone now?

Fixes: #216
commit 7279b24c (2021-11-01): ravedude: Cleanly terminate on CTRL+C

Install a CTRL+C handler to terminate with exit code 0 instead of the
default handler which exits with a failure.

Fixes: #216
closed 2021-11-01 by Rahix