Rahix/avr-hal #219: 8MHz ATMEGA328

IamTheCarl, 2021-09-09:
I created my own bootleg Arduino compatible devices using an ATMEGA328 (notice the distinct lack of p).
Getting this library to run on that device was fairly trivial ~~although ravrdude will need to be modified to support this use case~~.

However, there was one thing that was difficult. To cut costs to the absolute minimum I use the chip's internal 8MHz clock. Because of this, everything ran at half speed.

To fix this problem I forked this repository and added the necessary feature gates to add my custom board to the library.
The reason I do not make a merge request is that I feel that a custom board like mine is outside of the scope for this library.

However, a way to use 8Mhz instead of 16MHz is something that should be considered.
My fork did add this as a feature gate: https://github.com/IamTheCarl/avr-hal
Perhaps with the right adjustments, this could be added to the library?

Rahix, 2021-09-13:
Hi,

it sounds like you're building a custom board that's just a bit "similar" to an Arduino, is that correct?  Usually, my recommendation for this is to not use `arduino-hal`, but the MCU-hal crate specifically.  For your case that would be `atmega-hal`.

The MCU-hal crate does not make any assumptions about the clock speed so you're free to choose as you like (you tell each peripheral about the clock-speed via the generic parameters).  If your clock speed is constant, I would suggest the following pattern in your application code:

```rust
pub type CoreClock = atmega_hal::clock::MHz8;
pub type I2c = atmega_hal::i2c::I2c<CoreClock>;
// etc.
```

On the other hand, `arduino-hal` is designed similar to the official C Arduino cores.  It makes decisions about certain things so the user does not need to care.  The clock speed is one of them right now.  We do this to have `arduino-hal` be an easy starting place for getting familiar with avr-rust.  Once you want to dive deeper, the MCU-hal crates are what's intended to be used.

IamTheCarl, 2021-09-18:
That is perfect thank you.
I'll set that up later.

closed 2021-09-18 by IamTheCarl

stappersg, 2021-09-18:
On Fri, Sep 17, 2021 at 05:31:19PM -0700, James Carl wrote:
> That is perfect thank you.
> I'll set that up later.

A better way to say "thank you" is reporting
how well the advice set-up works