Rahix/avr-hal #222: Unable to print serial output on Adruino leonardo

timwillett4, 2021-09-15:
I am hitting an error when trying to print serial output using an adruino leonardo connected to a WIndows 10 PC.

thread '<unnamed>' panicked at 'assertion failed: e.kind() == std::io::ErrorKind::TimedOut', C:\Users\timwi\.cargo\registry\src\github.com-1ecc6299db9ec823\ravedude-0.1.3\src\console.r
s:24:17
stack backtrace:
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

I tested with Adruino IDE and it works fine there so it is noting with the USB Connection/Hardware.

**config.toml:**
[build]
target = "avr-specs/avr-atmega32u4.json"

[target.'cfg(target_arch = "avr")']
runner = "ravedude leonardo --open-console --baudrate 9600"

[unstable]
build-std = ["core"]

**.Full console output:**
cargo run
...
Writing | ################################################## | 100% 0.17s

avrdude: 1820 bytes of flash written
avrdude: verifying flash memory against target\avr-atmega32u4\debug\hello-world-arduino.elf:
avrdude: load data flash data from input file target\avr-atmega32u4\debug\hello-world-arduino.elf:
avrdude: input file target\avr-atmega32u4\debug\hello-world-arduino.elf contains 1820 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.04s

avrdude: verifying ...
avrdude: 1820 bytes of flash verified

avrdude: safemode: Fuses OK (E:CB, H:D8, L:FF)

avrdude done.  Thank you.

  Programmed target\avr-atmega32u4\debug\hello-world-arduino.elf
     Console COM6 at 9600 baud
thread '<unnamed>' panicked at 'assertion failed: e.kind() == std::io::ErrorKind::TimedOut', C:\Users\timwi\.cargo\registry\src\github.com-1ecc6299db9ec823\ravedude-0.1.3\src\console.r
s:24:17
stack backtrace:
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

Rahix, 2021-09-15:
Hi,

this is to be expected because the Arduino Leonardo, unlike the Uno, does not have a dedicated USART-to-USB converter chip.  This means you'll need an external USB-Serial converter, connected to pins D0 and D1.

You're probably wondering how the official Arduino software gets around that:  What they're doing is implementing a USB-Serial peripheral directly on the ATmega32U4 chip on your Leonardo.  We plan to do the same in `avr-hal` at some point, but this is not yet done (see issue #40).

timwillett4, 2021-09-15:
Okay thanks so much for the help.

closed 2021-09-16 by Rahix